prusa3d/PrusaSlicer #5937: Edit Menu Shortcut strings are not aligned

daniel-starke, 2021-01-30:
### Version
PrusaSlicer Version 2.3.0+win64

### Operating system type + version
Windows 7 (build 7601, Service Pack 1), 64-bit edition

### Behavior
Shortcut descriptions in menu items of `Edit` are not aligned to the right (except for `Search Ctrl+F`).

[image]

daniel-starke, 2021-01-30:
Does also apply to the `View` menu.

combolek, 2021-02-03:
This could be Windows 7 specific. Looks OK to me on Windows 10:

[image]

daniel-starke, 2021-02-03:
Looking at your screenshot it seems that the broken items on Windows 7 have a space between `+` and the addtional key on Windows 10. Maybe related to the issue?

YuSanka, 2021-02-05:
@daniel-starke Thanks for your report. I added some space symbols before shortcuts for these menu items. Please, retest [this build](https://drive.google.com/file/d/1KPpJBdYypwNWTyiOUDfE7RNJSlczCweu/view?usp=sharing)

daniel-starke, 2021-02-05:
Thank you, but the issue still remains.
[image]

YuSanka, 2021-02-05:
Can I ask you what Windows localization do you have?

daniel-starke, 2021-02-05:
German OS with Japanese localization.

bubnikv, 2021-02-06:
I am running the same Windows 7 build, actually it is Windows 7 32 bit,
English. I don't see the effects you are seeing.

Let's keep this issue open, if somebody else reports this issue.

daniel-starke, 2021-02-06:
I did a quick check on a different computer with native Japanese OS. Looks similar.
Microsoft Windows 7 Home Premium x64
6.1.7601 Service Pack 1 Build 7601 (6.1.7601.24545)

[image]

closed 2026-09-29 by prusaslicer-bot[bot]