prusa3d/PrusaSlicer #5938: Feature Request - Overhang Perimeter (Shell) Detection and

bradvaughan, 2021-01-30:
### Version
2.3.0+win64

### Operating system type + version
N/A

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
N/A (feature doesn't exist)

* _Describe the problem_
The existing combination of perimeter and bridging settings (i.e., print speed, fan speed, flow rate, etc.) is inadequate to achieve reasonable print speeds while maintaining quality on overhanging perimeters (e.g., the bow of 3DBenchy or other similarly aggressive overhangs).  The user must make a choice to print all external perimeters very slowly if they desire overhanging perimeters to be printed slowly.  There is no separate fan speed setting for this feature type, which can cause issues for cooling-sensitive materials such as PETG.  Lastly, there is no setting to adjust extrusion width or flowrate/multiplier for this feature type.

* _Steps needed to reproduce the problem_
Slice/Print 3DBenchy (or any model with similarly aggressive overhangs) at somewhat high speeds and notice degraded print quality on the overhangs due to external perimeters curling.

* _Expected Results_
Enhance or replace existing "Detect Bridging Perimeters" functionality to support the following:
1. Set overhang detection angle, separate from the existing "Bridging Angle" setting
2. Adjust print speed for this feature type separate from existing features
3. Adjust fan speed (or have it match existing "Bridges Fan Speed" setting for this feature type separate from existing features
4. Adjust extrusion width for this feature type separate from existing features
5. Adjust flowrate for this feature type separate from existing features

* _Actual Results_
N/A (feature doesn't exist)

_Is this a new feature request?_
Yes

bradvaughan, 2021-01-30:
Full disclosure, this is similar functionality to ideaMaker's "Overhang Shell Detection" functionality

arbego, 2021-02-28:
Please make this happen.

Profkanz, 2021-03-07:
Desperate for this additional functionality.

bradvaughan, 2021-03-12:
> I also noticed this, PrusaSlicer is not able to detect most overhangs currently.

I experience the same thing.  For instance, I can't seem to get it to detect the bow of a Benchy as an overhanging perimeter regardless of what I set the bridging angle to.

neophyl, 2021-03-12:
Bridging angle has nothing to do with detecting a bridge, its just the angle at which it prints bridges that have been detected.  Besides an overhang and a bridge are distinctly different things.  One is connected on one side, the other is connected at each end.

bradvaughan, 2021-03-12:
> Bridging angle has nothing to do with detecting a bridge, its just the angle at which it prints bridges that have been detected. Besides an overhang and a bridge are distinctly different things. One is connected on one side, the other is connected at each end.

You're correct.  I meant "Overhand Threshold".  I'll edit.

drpain, 2021-04-25:
Yes please!

halfshavedyaks, 2021-09-06:
+1 I would also very much like this feature. I currently have to choose between layer adhesion for a strong part, and tidy overhangs. The fan speed adjustment for overhangs is more important than the print speed adjustment, in my view, but both would be good.

quite a lot of these functions are in this fork of Cura:
https://github.com/smartavionics/Cura
however the overhang detection features are still buggy.

superDIYer, 2022-12-16:
Another +1 on this, please. Roughness in the overhanging outer perimeters of a model with chamfers or fillets against the print surface are my #1 difficulty.

Sverto, 2023-05-02:
I'm having a similar experience with overhangs (in the latest version beta alpha6).
While the new dynamic overhang speed settings help, they do not turn on the fan. 
Only the setting "Detect bridging Perimeters" does this. But it is too conservative in my experience.

Allowing the user to set the bridge detection angle or alternatively setting a dynamic fan speed would likely resolve this (in my situation).

closed 2026-09-29 by prusaslicer-bot[bot]