prusa3d/PrusaSlicer #5940: XY compensation setting for modifiers

sotosalex, 2021-01-31:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 1909

### 3D printer brand / version + firmware version (if known)
AM8 - Marlin 2.0.7

### Behavior
Add XY size compensation at the available settings of modifier blocks. 
An example case that this would be useful would be with gears. The user would be able to modify the diameter of the axis hole in the middle of the gear using a cylindrical modifier, while the dimensions of the rest of the gear remain unchanged.  

_Is this a new feature request?_
Yes

tokolist, 2022-04-15:
It will be also extremely useful for magnet inserts and screw holes adjusting. If you apply xy compensation to entire model, this might mess it up, for instance if there are joints with some tolerances.

ubqtous, 2022-06-12:
Love this idea! I just ran into having issue of having to need per model right now and can't print by batches anymore

jmding8, 2023-05-17:
would also love this feature for a model I'm trying to print

3Dietrich, 2024-03-03:
This topic came up for me today regarding the XL multi-tool head: I have an SVG (symbol) on the main object, which I scaled down so it would fit in size. However, the lines became too thin. I would like to expand the XY dimensions of this part only, without affecting everything else.

tokolist, 2024-03-03:
> This topic came up for me today regarding the XL multi-tool head: I have an SVG (symbol) on the main object, which I scaled down so it would fit in size. However, some lines of the symbol became too thin, so I would like to expand the XY dimensions of this part only, without affecting everything else.

I ended up using Blender for cases like this, but if this feature comes in slicers, that would make things so much easier. Orca Slicer already supports XY compensation for holes and contours separately. I hope we will get XY compensation in slicer modifiers soon too.

tjc2005, 2025-04-09:
Cmon we still don't have this 3 years later?

ulayera, 2025-06-17:
Disappointed this is not a feature already

domdfcoding, 2026-09-03:
I still think this is a worthwhile feature to have.