prusa3d/PrusaSlicer #5947: Artifact in seam with 0.6 printer profile

pahofmann, 2021-02-01:
### Version
2.3.0+linux-x64-GTK3

### Operating system type + version
POP OS 20.10

### 3D printer brand / version + firmware version (if known)
Prusa mk3s+ (Firmware 3.9.3)

### Behavior
I was printing a model just fine with stock 0.4 Nozzle and stock settings/profiles.
After switching to 0.6 nozzle and stock settings/profiles I have an artifact, always at the same spot. I printed the same gcode multiple times, and resliced the file (even in different sizes) and the artifact is always there. The smaller the model the bigger the artifact.

I understand that the seam is more pronounced with a 0.6 nozzle. But a part of the seam is offset when slicing with the 0.6 profile and that offset part is a lot deeper (1-2mm)  then the rest of the seam in the printed object:

Model with 16cm height (~1cm long Seam Offset that is 2mm deep):
[image]

Model with 20cm height (~3mm long seam offset that is 2mm deep):
[image]

The offset part of the seam can be seen in the slicer, but the rest looks fine and I don't see a reason why the seam should be deeper there:
[image]

#### Project File (.3MF) where problem occurs
[Dried-Flower-Circle-Vase.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/5904521/Dried-Flower-Circle-Vase.amf.zip)

rtyr, 2021-05-05:
> After switching to 0.6 nozzle and stock settings/profiles I have an artifact, always at the same spot.

You mentioned stock settings/profiles, but I can see your print settings are not stock. You have "Only retract when crossing perimeters" option enabled, which can introduce issues like this. 

Have you tried it with stock profile?

pahofmann, 2021-05-05:
Sorry maybe I uploaded the wrong zip. I tried many different settings including all stock. 

I used the seam paint feature to get the seam in line with the rest of the seam and now the model looks fine.

rtyr, 2022-05-17:
This should be fixed since PS2.4.0, where we improved seam placement. Inner seams should be close to outer seams now, even if seam painting feature is used.

closed 2022-05-17 by rtyr

pahofmann, 2022-05-17:
Thanks I will check it out. Unfortunalty it seems something has changed with the seams or seam painting aswell.
If I slice the same files (3mf where seam painting is saved) now with Prusa Slicer 2.4 the seams are in a different position.