prusa3d/PrusaSlicer #5949: PrusaSlicer 2.3.0+win64 crashes when opening file picker when specific file is in folder

flexd, 2021-02-01:
### Version
Version 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 34,295MB

### 3D printer brand / version + firmware version (if known)
Ender 3 v1 (modified, SKR 1.3, TMC2209, direct drive, Klipper). Not relevant for this problem

### Behavior
* _Describe the problem_
I have a file from here https://www.thingiverse.com/thing:3270228/files, "Frame_X_1.3.STL" (815kb), that will cause the File Picker to crash every time. If I open the file in PrusaSlicer from Windows File Explorer, it loads fine and all is well. But if I try to click on the Add Button (shortcut Ctrl-I) in 3D Editor view, the whole application hangs forever "Not Responding". As soon as I delete that file, the problem is gone.

(Also attached here, zipped because GitHub does not support stl files as uploads)

[Frame_X_1.3.zip](https://github.com/prusa3d/PrusaSlicer/files/5904710/Frame_X_1.3.zip)

* _Steps needed to reproduce the problem_
1. Download the file mentioned either from Thingiverse or the zip attached.
2. Open file directly with PrusaSlicer to confirm it actually does load fine (seemingly).
3. Try clicking "Add" to add that file. 
4. Observe that PrusaSlicer actually hangs.
* _Expected Results_
PrusaSlicer should not become unresponsive.
* _Actual Results_
before opening file (item was added just fine)
[image]
Trying to click Add
[image]
File opened directly from File Explorer (slices fine and everything)
[image]
Trying to add a file when the weird Frame_X_1.3.stl file is not present (deleted)
[image]

_Is this a new feature request?_
No

combolek, 2021-02-03:
FWIW, I tried to reproduce it and couldn't. It all worked fine for me.

closed 2021-09-22 by bubnikv

flexd, 2021-09-22:
I do not have an issue with my system, it was only that file with that version. I have not had a problem after updating or going back to another version. I'm not by that computer right now so I can't tell you what version. Feel free to close, I won't be able to replicate it now.