prusa3d/PrusaSlicer #5950: Ironing all solid surfaces stays on when "Enable Ironing" is unchecked

bakerjvb, 2021-02-01:
### Version
2.3.0+Win 64

### Operating system type + version
Windows 10
Build 190421.746

### 3D printer brand / version + firmware version (if known)
Prusa I3 3.93
### Behaviour
Switched ironing on
    Ironing type: all solid surfaces. 
I sliced and built a few models with this to explore its effect. Now:
Uncheck ironing
Slicer still irons each surface!

* _Expected Results_
 On uncheck "Enable ironing" Ironing should not take place
* _Actual Results_
 Ironing takes place
And results in freezing of the head as it is travelling very quickly with no filament flowing.

a) If "Enable Ironing" is unchecked, please remove all ironing actions.
b) Please add an option to alter fan speed to xx% only when Ironing is taking place.
c) Please add a feature to manage the xx% speed of the heatsink fan whilst ironing is taking place.

_Is this a new feature request?_ a: no.  
but B and C are enhancement requests.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) I tried but:
Error on upload: "we don't support this file type." .3mf is not a supported file type to upload

bakerjvb, 2021-02-04:
my apol;ogies. I reinstalled v2.2.0 and this "ironing" still happens.
The part of a single layer goes down, then the head is moved over that part of the layer again; with no filament extruded.

Can someone tell me how to switch this off?

fe60, 2021-02-07:
Your comment confuses me. You switched back to PrusaSlicer 2.2 and ironing still happens?
Ironing has been introduced in PrusaSlicer 2.3 only.
https://help.prusa3d.com/en/article/ironing_177488 (first sentence)

rtyr, 2021-03-04:
I don't understand. Please provide your 3MF project file (you have to zip it first).

bakerjvb, 2021-03-26:
Right, Having switched back to V2.2 and forward again to V2.3 I am convinced that this issue doesn't have anything to do with ironing. However it appears that an **ironing effect** is taking place at various points around all prints.
The ironing effect here stops filament flow, and repeats the last set of movements whilst staying at the same Z.  
(on a big model with a long set of movements, this causes freezing of the chambef even whemn printing PET-G at 265degrees)
The additional effect is to substantially lengthen the build time.
How can I remove this effect? 
I attach an example of an amf file demonstrating this effect.
[crossmember crossection.zip](https://github.com/prusa3d/PrusaSlicer/files/6212507/crossmember.crossection.zip)

bakerjvb, 2021-03-31:
This is actually an issue with the Wipe on Retract code and I have raised a separate issue and am closing this issue now.

closed 2021-03-31 by bakerjvb