prusa3d/PrusaSlicer #5951: Feature Request - Custom GCode at specific GCode Position

ggrote, 2021-02-01:
It would be nice to have the ability to insert custom GCode at a specific GCode position not only at a specific Layer Height.
This way it would be possible to select the exact moment for the pause, which is very useful for inserting magnets or other stuff in the print after the outer walls have already been printed.
[image]

fe60, 2021-02-07:
What is the problem you want to solve with this feature?
From my experience for the insertion of magnets or nuts only the layer matters.

ggrote, 2021-02-08:
The problem is that pausing at a layer height pauses before the outer walls are printed.
If you resume the print after inserting magnets at first the outer walls are printed, but the first centimeters are not of good quality.
So it would be better to move this first centimeters after the pause into the infill or something other before the magnet holes are closed.

fe60, 2021-02-08:
For objects with infill adding "Infill before perimeters" to the settings offered for modifiers, could be a solution.

But the object shown in your initial post has no infill. So I guess you would stop the print while it's printing one of the internal perimeters to get the printer continuing there after the pause.
By default the first thing printed starting a new layer, are the inner perimeters. Having this in mind, I don't know how stopping the print within a layer could make a difference.

rtyr, 2021-03-04:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5195

Devilscave, 2025-03-30:
I would like to join here, as I was referred to this one in my issue [#14364.](https://github.com/prusa3d/PrusaSlicer/issues/14364)

I would also like to have this function to add a simple color change within a layer. As this color function only becomes active with an MMU. So it would also work very simply for MonoColor

[image]

bkw777, 2025-06-03:
I want to be able to pause at a specific point to paint the top surface of some support with sharpie before the non-support part of the layer starts drawing.

It's a very steep overhang for a few layers at the bottom of a cylinder laying on it's side. It's impossible to marker just the support without touching the part after both the part and support parts of the same layer have been printed right next to each other.

There is no amount of xy seperation that works, I want the support to be about a nozzle width away from the part in xy, and I want to ink the support top and inside after the support is drawn but before the part draws.

The layer boundary does not allow this, but I can see exactly a spot in the gcode that does provide this. Being able to insert a pause or other custom gcode from the left-right slider just like the vertical slider would do it.

closed 2026-09-29 by prusaslicer-bot[bot]