prusa3d/PrusaSlicer #5952: Detractions not showing in toolpath

n9jcv, 2021-02-01:
prusaslicer 2.3.0 release

slice the xyz calibration cube
preview the tool path
select option to view retractions and detractions
retractions show as pink/purple   detractions should show as blue, but do not show
if you turn off retractions, then you see detractions, but when both selected it does not show detractions
[image]

n9jcv, 2021-02-01:
Where it is outer perimiter, it retracts after the wipe on the front face (x) of the cube, then moves to the blue infill, at the start of the blue infill is a detraction, but it does not show.  If you turn off retractions then the detractions show.

n9jcv, 2021-02-01:
Version 2.3.0+win64

rtyr, 2021-02-02:
This should be already fixed in current master. I can retest with your 3MF project, if you can provide it.

n9jcv, 2021-02-04:
[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5923200/xyzCalibration_cube.zip)
simple calibration cube, I use stl and not 3mf

rtyr, 2021-02-04:
Works fine in current dev version. The fix will be included in the next PrusaSlicer release. Closing.

closed 2021-02-04 by rtyr