prusa3d/PrusaSlicer #5955: transfer settings between Ender 3 pro and prusa printer

WimmerDuiven, 2021-02-02:
[image]
[image]

2.3.0 Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Prusa  3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* Prusa slicer
* select ender 3 pro
* store object
* open prusa slicer
* select stored file (mentioned above)
* Change printer to prusa printer
* copy settings
Ender 3 pro prints startline standard on the left on the bed.
Prusa start line standard at the top,
Now the  Prusa prints the line on the left See pictures.

* change to 
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-02-02:
I am not sure if I understand you correctly, but it seems like correct behavior. 

You should either use the "Discard" button or uncheck the settings which you don't want to transfer to newly selected profile. The intro line is defined in the start g-code. I think you should just use the "Discard" button in this particular case.

[image]

WimmerDuiven, 2021-02-02:
First Time I posted something, 
Do I understand well, that If  you transfer the settings between ender3 pro and the Prusa  you also tranfer the code for the first line?
I am not familiar with G code yet (I am retirred 71 time enough to learn)

rtyr, 2021-02-02:
Yes. This dialog appears if you are switching from the preset, which has some unsaved changes (from the original state). You have option to save those changes, transfer them to the profile you are switching to or discard those changes and just switch to different preset/printer. I suppose you don't want to tranfer any printer settings from Ender to Prusa i3, so you should use option "Discard". You can also tick "Remember my choice" to avoid this dialog in the future.

WimmerDuiven, 2021-02-02:
Thanks a lot

closed 2021-02-02 by WimmerDuiven