prusa3d/PrusaSlicer #5956: Why is changing of flow in the default Custom G-Code?

rw3iss, 2021-02-02:
PrusaSlicer 2.3.0, latest firmware...

I kept noticing the printer settings showing 'Flow: 95%', even though all my flows are set to 105% or so...
I looked at the custom G-code and I see this line

`M221 S{if layer_height<0.075}100{else}95{endif}`

Why is this even in here, and not in the custom settings? Does this override the slicer's user-configured flow settings (it seems like it does? As the printer shows Flow: 95%), or is this firmware/G-Code value some factor of the configured custom user-configured flow?

rtyr, 2021-02-02:
It is explained here https://github.com/prusa3d/PrusaSlicer/issues/1446#issuecomment-443405690. Closing.

closed 2021-02-02 by rtyr

rw3iss, 2021-02-02:
Ah, ok, thanks.