prusa3d/PrusaSlicer #5958: Improved retractation-detractation preview

choubisky, 2021-02-03:
### Version
_Version of PrusaSlicer : 2.3.0

_Use `About->About PrusaSlicer` for release versions_

### Operating system type + version
Win10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S + 3.9.2

### Behavior
* _Describe the problem : 
would like to know if there is a possibility of better rendering in Prusaslicer to detect perimeter overlaps.
indeed, when cutting a part, PrusaSlicer did not show me any particular rendering problem but once the part was printed,
artefacts appeared on the wall which was supposed to be smooth I think that by printing the inner perimeter,
the nozzle has come to overlap the other perimeter and created an extra thickness.

My print settings were with a 0.4 nozzle and 0.3mm height in PLA. The thickness of the recess inside the part should be 0.8mm.
 
I understood where my problem was but While chatting with a friend he showed me that on Simplify3D the rendering was different and showed exactly
how the part was going to be printed (CURA also doesn't show the problem)

[Test pièce 1.zip](https://github.com/prusa3d/PrusaSlicer/files/5916829/Test.piece.1.zip)
[image]
[image]
[image]

  * _If this is a command-line slicing issue, include the options used_
  My print settings were with a 0.4 nozzle and 0.3mm height in PLA. The thickness of the recess inside the part should be 0.8mm

_Is this a new feature request?_
As you can see the rendering on Prussaslicer looks good but once I printed I saw some extra thickness on the smooth side of the part. It's an extrusion width issue I think and found the solution but I wish Prusaslicer had shown me exactly how the part was going to print rather than finding out on print. By comparing with Simplify3D, this shows exactly how the part will print so we can act accordingly, modify the settings, modify the part ... etc.

rtyr, 2021-02-03:
This print "defect" is most likely caused by non optimal extrusion width in relation to the thin wall thickness of your model (there is some overlap of the perimeters in that problematic area). Whether it will be visible on the printed model or not depends on particular material, temperature, etc. There is certainly room for improvement when slicing such objects (see the issues with "gap fill" label), but the preview is OK. The preview in S3D is not better, those points are just retraction points, it doesn't show the defect you are reffering to (it doesn't even match witch your photo if you look closely).

choubisky, 2021-02-03:
Thank you for your reply. Indeed by looking for the cause of the problem, I noticed that with the 0.3 profile and a 0.4 nozzle, the extrusion width was 0.5 so as the thickness at this location of the print was 0.8mm, with the passage of the nozzle, there is an overlap of the perimeters. But it must be able to be seen with the slicer? (I used Prusament GalaxyBlack) because it's a nozzle extrusion width issue and not a filament and / or temperature issue ... etc?

choubisky, 2021-02-03:
I modified the parameters of "extrusion width" to put 0mm to let Prusaslicer manage this parameter and that to attenuate the phenomenon almost completely. Only "holes" remain in the surface

[image]

[image]

[Test pièce 1_V2.zip](https://github.com/prusa3d/PrusaSlicer/files/5918690/Test.piece.1_V2.zip)

closed 2026-09-29 by prusaslicer-bot[bot]