prusa3d/PrusaSlicer #5966: It would be nice to not have "painted supports" areas removed when moving/rotating (possibly scaling) a mesh

rw3iss, 2021-02-04:
PrusaSlicer 2.3.0
Ubuntu 20.10
Prusa MK3S

I have a model where I want to paint supports underneath only one area, on an inner section, however, I can't zoom into the model because there is another wall opposite from it. I instead flipped the model over 180 degrees, and then used the 'Clipping of view' feature (which is a great feature!) on the "Paint-on supports" tool, and then was able to paint underneath the overhang there. The issue is that when I flipped the model back over, the painted supports disappeared :( I don't see any other way to put manual supports there, so am requesting this ability. Please see the picture for the area I mention.
[image]

rw3iss, 2021-02-04:
Err... nevermind, it looks like the painted areas are still there, but for some reason the supports aren't being generated. Please see screenshots. These setting should be correct?

Actually I'm noticing it keeps overriding the settings... If I select 'Supports: Everywhere', then it turns on 'Auto-generate supports', but I don't want it to. I just want to generate supports where I painted (which should generate them on the model). 
If I turn off 'auto-generate', it changes the Supports setting backing to 'For support enforcers only'. Neither scenario generates supports under the painted area. Is this normal behavior? Please see screenshots.
[image]
[image]
[image]
[image]

rw3iss, 2021-02-04:
Here is the .3mf file of the project:
https://drive.google.com/file/d/15UTFzYqSlqO6gpSF_KaQeXwzBavuYmuH/view?usp=sharing

The model is manifold, no errors. I really don't understand why I can't get supports to be generated under the inner overhangs. I've tried all manner of settings. Is this a limitation of the software, or a bug?

rtyr, 2021-02-05:
Disable "Don't support bridges" if you want supports there.

[image]

Non-related side note. 
[image]

rw3iss, 2021-02-05:
Ahh!! Thank you, again, @rtyr !!

Noted on the extrusion width as a percentage of the layer height... that is not obvious! Thanks.  :)

closed 2021-02-05 by rw3iss