prusa3d/PrusaSlicer #5969: Printing pb because gcode seems not good

zergeniel, 2021-02-04:
### Version
_Version 2.3.0+win64_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 latest update#s_
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18362), 64-bit edition
Total RAM size [MB]: 4,178MB
OpenGL installation
GL version:   4.6.0 - Build 26.20.100.7156
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 510
GLSL version: 4.60 - Build 26.20.100.7156
Installed extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_robustness_isolation
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_compiled_vertex_array
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_object
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_polygon_offset_clamp
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_compression_s3tc
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_IBM_texture_mirrored_repeat
GL_INTEL_conservative_rasterization
GL_INTEL_fragment_shader_ordering
GL_INTEL_framebuffer_CMAA
GL_INTEL_map_texture
GL_INTEL_multi_rate_fragment_shader
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
_Artillery sidewinder X1 bought at 3dexpert.com / version 4.1 stock, i don't know the firmware version #s_

### Behavior
* _When i want to print an objet vase/spiral demo cube for example. My init gcode is OK, i used it with cura and S3D without error and defuncts. The print start with a first uncomplete perimeter. The printer print 2 sides of the cube, then go to the start of the print diagonally and continue the print by filling the first layer over this line... This is not present in the slicer on the first layer shown by prusaslicer in preview mode (all seem ok). I tried many parameters change, copy some from web etc. always this weird bad start _
I put all configs and a gcode sample in the zip  file (picture, config.ini, 3mf, amf, etc.) with some pictures of the actual result plus the expected one out of the zip file (sorry, last picture made after upload of the zip file).
I really prefer Prusa interface between cura and simplify3d (too expensive, i just pay for a few days and asked a refund because of the non user friendly interface). I printed more than 20 tries with a lot of different combination of parameters...
[image]

I hope my description is clear because english's not my natural language.
* 
[image]

* _Steps needed to reproduce the problem_
* I don't have to, it's always happening and don't know why.
  * _If this is a command-line slicing issue, 

include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
In the zip file

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF
[Support Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/5928305/Support.Prusa.zip)
) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

stay safe,
Ludovic

PS : I hope you will find the problem (perhaps it's me, i hope not) because i really don't like Cura and S3D is too expensive.

rtyr, 2021-02-05:
- That diagonal line is not present in the gcode and the perimeter is also OK in the gcode. It may be caused by corrupted filesystem on your flash drive. You can try to format it or just use different one.

Some other observations:

- At the beginning, you are retracting 9.5mm and deretracting only 6.5mm (3mm in your custom g-code + 6.5mm from the settings)
- You are using 0.35mm layer height and 0.7mm extrusion width for the first layer. I am not sure if it is intended.

zergeniel, 2021-02-05:
Hi,
 thx for your quick answer, you're right it's a mistake, i will correct it by removing the retraction in the starting gcode.
Normally, i'm using a 0.35mm height first layer then 0.07 (usually 0.12 for detailled objects) or more until 0.2/0.3. If the first layer is wrong, that may explain the pb. It should be the opposite => first layer, slower, warmer and thicker than the others theorically no ?

I'll format the usb key and try an other one or a 16/32 Mo sd card (i think my usb keys are too big (more than 32 Mo) for the printer.

PS : I received my printer only 3 week ago, so i'm doing mistake, sry.

----- Mail original -----

rtyr, 2021-02-05:
On the second look, I believe I now understand, what is going on.

1st issue - part of the perimeters is not printed because of that retraction/deretraction issue in your gcode (9.5mm vs 6.5mm)
2nd issue - diagonal line. It is caused by your setting "Only retract when crossing perimeters". It basically means it freely oozes the filament when travelling inside the part.

So it should be OK, after you fix the start g-code and disable the "Only retract when crossing perimeters" option.

This is causing that diagonal line.
[image]

zergeniel, 2021-02-05:
I'll try these corrections this evening when the actual print will finish. Thx for your time and advices. I'll tell you if it's ok. ;-) Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : rtyr <notifications@github.com> Date : 05/02/2021  11:36  (GMT+01:00) À : prusa3d/PrusaSlicer <PrusaSlicer@noreply.github.com> Cc : zergeniel <zergeniel@free.fr>, Author <author@noreply.github.com> Objet : Re: [prusa3d/PrusaSlicer] Printing pb because gcode seems not good (#5969) 
On the second look, I believe I now understand, what is going on.
1st issue - part of the perimeters is not printed because of that retraction/deretraction issue in your gcode (9.5mm vs 6.5mm)
2nd issue - diagonal line. It is caused by your setting "Only retract when crossing perimeters". It basically means it freely oozes the filament when travelling inside the part.
So it should be OK, after you fix the start g-code and disable the "Only retract when crossing perimeters" option.
This is causing that diagonal line.

zergeniel, 2021-02-05:
Hi,
   here the result of the print with the 2 modification you said (in starting Gcode and unchecked option "only retract ... perimeters")... it's not ok, always the line and the first perimeter not printed and the infill line width not very large (should be 0.6).

The picture, the generated gcode, the stl file (just in case) and the parameters in ini file joined.

----- Mail original -----

2nd

rtyr, 2021-02-05:
Please attach your gcode and 3mf project. I will do quick check. But if the gcode will be fine, I will likely close this issue.

zergeniel, 2021-02-05:
Hi,
   the gcode was attached in the previous mail but here it is again with the 3mf file. When i print with cura, i have other problematic but not this diagonal line...

----- Mail original -----

rtyr, 2021-02-05:
It is not attached, you need to use web interface, not email.

zergeniel, 2021-02-05:
[image]
[Support Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/5935310/Support.Prusa.zip)

Ok i send this one by github.

rtyr, 2021-02-05:
The provided g-code is OK. I also printed it (see photo). The output from PrusaSlicer is OK.

[image]

zergeniel, 2021-02-05:
[image]
[image]

So you think it's the printer marlin's code that have a bug ?

zergeniel, 2021-02-05:
When i generate the code with cura, or simplify, i don't have this line, is there a gcode command used by prusa that is misunderstood by the printer ?

rtyr, 2021-02-05:
The gcode is fine. Contact Artillery customer support.

zergeniel, 2021-02-05:
I will open an issue on 3dexpert about this problem, on wich printer did you print my gcode on the picture please ?

rtyr, 2021-02-05:
Custom printer with Marlin 2. I am closing this issue. Feel free to add additional info, when you resolve this problem.

closed 2021-02-05 by rtyr

zergeniel, 2021-02-06:
Ok, thx for your time.

rtyr, 2021-02-10:
It seems, the g-code from PS is working fine for other X1 owners, judging from this pull request https://github.com/prusa3d/PrusaSlicer-settings/pull/118.

SzabolcsHornyak, 2021-02-10:
@rtyr  Which G-Code exactly?

rtyr, 2021-02-10:
It is attached in this issue.