prusa3d/PrusaSlicer #5970: Feature request: paint seam lines perfectly vertically

DaveTSG, 2021-02-04:
### Version
_2.3.0_

### Operating system type + version
_MacOS Big Sur (v11.2)_

### 3D printer brand / version + firmware version (if known)
_Stock Prusa i3 Mk3S with firmware v3.9.1, but the printer and firmware are irrelevant anyway..._

### Behavior
* _Using the new paint seams feature in PrusaSlicer v2.3.0, it is impossible to draw a straight line with the mouse across large models, so seams end up not straight. See below screenshots._
* _Paint a seam line with your mouse. For any large model, it will be impossible to make the painted seam region straight._
* _Two possible solutions: a) allow the user to click to generate two points, and PrusaSlicer would automatically create the line between the two points, or b) allow the user to click in one point (e.g. at the top of the model) and PrusaSlicer would automatically create a straight/vertical seam line from that point all the way down to the print bed (or perhaps until the next point, as that would then allow the user to specify multiple seam positions for different z-axis ranges). _
* _Yes, it is still possible to simply not paint seams at all and let PrusaSlice do it automatically, but then the seams don't end up where they're needed..._
* _The [seam position options](https://help.prusa3d.com/en/article/seam-position_151069) (particularly "nearest") help minimise the effect but it's not perfect._

Screenshot of model with painted seams:
[image]

Screenshot of sliced model showing resulting seams here:
[image]

_Is this a new feature request?_
Yep.

#### Project File (.3MF) where problem occurs
PrusaSlicer Project File (.3MF) [here](https://www.dropbox.com/s/iq8mmktyjgn020m/test1.3mf.zip?dl=0).

N.B.: there is a work-around available in MacOS by using the operating system's accessibility features to use the keyboard to move the mouse cursor (discussion [here](https://www.facebook.com/groups/Prusai3users/permalink/1720280244823116/?__cft__[0]=AZXt78hQ1moqH0N75bLW0Fqxdc2Oh-5IZKvHwRzh-2z-WnQJLqIge-I5t0hjZxzgeP65rI0GCYtjAmdHVV8MmWP8utgCf9wL1pDbUFpFt2L9CnTL5kHChNb3_8SIXH9BHbjfuw_04wfZP4F9BFfrsoZt&__tn__=%2CO%2CP-R)), but this is cumbersome and only works in MacOS. It would be best if this feature was implemented in PrusaSlicer directly for all operating systems.

Thanks for considering this request.

Area5142, 2021-02-04:
Just a suggestion.
When drawing the seam, direction could be locked to vertical (z) moves only by pressing Ctrl or Shift while dragging the mouse. This option is simple to use and enable locking to vertical whenever it is needed in drawing the seam.
This method is used by other drawing programs.

DaveTSG, 2021-02-04:
> Just a suggestion.
> When drawing the seam, direction could be locked to vertical (z) moves only by pressing Ctrl or Shift while dragging the mouse. This option is simple to use and enable locking to vertical whenever it is needed in drawing the seam.
> This method is used by other drawing programs.

Agreed, that would also be a good idea. Probably the best idea so far, as it is then consistent with other software...

bttguy, 2021-02-05:
Just wanted to open a FR for this. It would be soo cool, since this seam painting is way better than any algorithm for seam contorl, just annoys me that it still print random inside the defined area. I have to mess around with the thinnest brush zoomed in max and play for a long time to get what I want. I am not sure how hard it would be to implement a fixed line, but right now it's a big miss. Trust you guys, I already switched from Cura, make it even more worthy :D
Also I'd prefer to visualise seam in a more apparent way to be able to tweak to perfection, but this one is for a differe.t FR I guess.

mark-hahn, 2021-07-28:
There is an easy way to get perfectly straight lines and also a way too make them vertical.  See the comments [here](https://help.prusa3d.com/en/article/seam-painting_168620/)

DaveTSG, 2021-07-28:
> There is an easy way to get perfectly straight lines and also a way too make them vertical. See the comments [here](https://help.prusa3d.com/en/article/seam-painting_168620/)

If you mean the keyboard/accessibility option, I’ve already explained why it’s not a good solution in my OP.

If you mean the other comment which details a bug that the user could exploit, relying on a bug is a terrible idea because it’ll stop working when the bug is fixed.

mark-hahn, 2021-07-28:
>   relying on a bug is a terrible idea because it’ll stop working when the
bug is fixed.

I don't understand that logic. This is a future problem, not something to
worry about now.  I'm going to keep using it as long as possible.
[image: image.gif]

AdrianGarside, 2023-03-27:
Glad the bug workaround still works. But really - over two years later and a simple and powerful (and very typical) drawing aid feature isn't implemented yet? :(

imh1kiko, 2023-06-12:
Bump? Still nothing on this?

Sveeeeeen, 2023-08-27:
Bump... over 2 years old and still missing.

jkotowicz1, 2023-08-30:
Bump... Even when using bug, you have to eyeball the second point of seam "line". Just let me paint ideal horizontal or vertical lines while holding Shift

jurassic73, 2024-01-28:
Bump - I am really missing this feature in the slicer.

animoose, 2024-07-30:
Adding another vote for this.

It also seems a bit poor to keep closing other bugs as duplicates of this one, when there is no apparent progress on it.

jurassic73, 2024-09-21:
This method can help in the mean time. It exploits a Windows feature for mouse use.

https://www.youtube.com/watch?v=9cUUqdCF1_o

DaveTSG, 2024-09-21:
> ### Operating system type + version
> _MacOS Big Sur (v11.2)_

> This method can help in the mean time. It exploits a Windows feature for mouse use.
> 
> https://www.youtube.com/watch?v=9cUUqdCF1_o

That's not going to help me for what should be obvious reasons.

dougemon, 2025-09-10:
Hello! Just wanted to add on to this thread, I really would love the ability to draw straight line seams or other types of paint on features.

iuvi7, 2025-11-15:
Vote +1 for add "vertical" line as drawing tool

ex-nerd, 2026-09-02:
Adding a comment here to keep this open. I've confirmed the feature is still missing from PrusaSlicer 3.

Also closing/referencing my own duplicate of this: https://github.com/prusa3d/PrusaSlicer/issues/12775