prusa3d/PrusaSlicer #5971: Error at slicing and gcode creation ( missing model details )

XboxOldie, 2021-02-05:
### Version
Version 2.3.0+win64_

### Operating system Win10 PRO + 20H2

_In case of 3D rendering issues, please attach the content of menu Help -> System Info PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,097MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.66
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1050/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 342 lines cut ...]
Ender 3 Pro

### Behavior
* Box model with a thin tab part between Bottom and Top.
* I can provide the Box model to slice.
* _Steps needed to reproduce the problem_
* After slicing the shown model is split into two parts and also print is done as 2 parts.
* _Expected Results_ would have been sliced and printed like the stl model.

[Prusa Error-Box mit Deckel zusammen 50 x50 v1.zip](https://github.com/prusa3d/PrusaSlicer/files/5929576/Prusa.Error-Box.mit.Deckel.zusammen.50.x50.v1.zip)

* _Actual Results_ Was split into 2 parts.
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
[image]

_Is this a new feature request?_yes

#### Project File (.3MF) where problem occurs

All files stl and 3mf attached

I did slicing the same stl in Cura for testing without any error like this.

Hope you can find the issue because I like to work with Prusa but the prints need to be correct and not like this.

neophyl, 2021-02-05:
No files are attached.  You need to zip them up for them to be accepted.
Without the 3mf its hard to be positive but looking at the picture it looks like that joining area is too thin to be sliced.  If you are using a default type profile then for a 0.4mm nozzle the extrusion width will be set to 0.45mm.  Anything less than that will not be rendered.  
You have a few options if that is the case.  
-You could enable detect thin walls which *may* then slice it (if its not already turned on).
-reduce your extrusion width slightly as that will pick up thinner features. Try 0.4mm to match your nozzle width first.

- you could also apply a xy compensation but in this case I don't think that will be a viable option as it grows everything in the x/y which will probably make that part not fit together (it looks like a case with a living hinge so has box and lid).

The best solution would be to make the hinge area slightly thicker in cad if you can as that way it will slice.

XboxOldie, 2021-02-05:
@neophyl 
I have already addes the zip file above but here it is again.
[Prusa Error-Box mit Deckel zusammen 50 x50 v1.zip](https://github.com/prusa3d/PrusaSlicer/files/5931316/Prusa.Error-Box.mit.Deckel.zusammen.50.x50.v1.zip)

Thin walls option was on.
The thickness of this joining area was designed with 0.6 mm which should be ok to be processed.

I loaded this also in Cura to check if something is wrong with my design and in Cura it was done perfectly.
So I think thats definitely a bug in Prusa slicer which needs to be solved.

jmd73, 2021-02-05:
@XboxOldie Remove the "detect thin walls" parameter and it slices ok.

XboxOldie, 2021-02-05:
@jmd73 
Thank you so much, after remove of the thin wall, slicing shows now the small join inbetween correctly.

This parameter should probably better explained because I was thinking it will improve printing in case of thin walls.
In future I will not switch it on anymore if surfaces are removed in this case.
Assume the print will be OK now.

Thanks again for the quick solving feedback, I will close this task now.

closed 2021-02-05 by XboxOldie