prusa3d/PrusaSlicer #5974: slicing fails or takes too long

rgiga, 2021-02-05:
### Version
Version 2.3.0+win64

### Operating system type + version
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,112MB
OpenGL installation
GL version:   4.6.0 NVIDIA 460.79
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1660 SUPER/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 335 lines cut ...]
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### 3D printer brand / version + firmware version (if known)
Prusa Mini, latest firmware

### Behavior
Model can't be sliced when oriented correctly, just stops on preparing infill. 

#### Project File (.3MF) where problem occurs
remember to place on face where embossing is, originally forward looking face

[loncek_dvojni.zip](https://github.com/prusa3d/PrusaSlicer/files/5933969/loncek_dvojni.zip)

foreachthing, 2021-02-07:
It sliced within a few seconds on my Windows 10 PC.

PrusaSlicer
Version:   2.3.0
Build:     PrusaSlicer-2.3.0+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,275MB
OpenGL installation
GL version:   4.6.0 NVIDIA 461.40
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

<details open>
<summary>System Info</summary>
<br>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 326 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

foreachthing, 2021-02-07:
Same here.

<details open>
<summary>System Info</summary>
<br>
PrusaSlicer
Version:   2.3.0
Build:     PrusaSlicer-2.3.0+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 34,275MB
OpenGL installation
GL version:   4.6.0 NVIDIA 461.40
Vendor:       NVIDIA Corporation
Renderer:     GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 326 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

rgiga, 2021-02-07:
Did you place it on bed like I've noted? Since I've had background slicing i couldn't save/export 3mf in correct orientation - slicer stopped responding to windows/mouse, but here is it again in correct orientation to be slice.

Here is the image how it looks at my slicer.

[image]

Here is the model in correct orentation

[loncek_dvojni correct orientation.zip](https://github.com/prusa3d/PrusaSlicer/files/5939402/loncek_dvojni.correct.orientation.zip)

foreachthing, 2021-02-07:
> Did you place it on bed like I've noted? 

At first, no but then yes. It just takes forever. Since there is no real progress indicator of what's going on (it just sits at "Preping infill"), I don't know whether it's doing something or if it's looping until infinity.
If you rotate the model 180°, it slices within seconds.... so .. wth?!
I let Craftware slice it and it only took two seconds (in the desired rotation).

jmd73, 2021-02-08:
I tried the 3mf and @foreachthing is right. Bridge detection seems to take "forever" at layer height 0.4.
I don't know why when part is rotated like that. But when checking object with Blender and FreeCad there seems to be errors in the object.

When repaired with Netfabb it sliced but it took a little while.

So the problem seems to be the "logo" on the bottom.

rgiga, 2021-02-08:
Yep the problem is embossing at the bottom, it looks like it's too complex, that's why i uploaded it here. I've tried some other slicers (cura, s3d) and they crunched it without any problems (original model, non 'repaired'). I guess repairing is mostly just simplifying vectors at the bottom, since 3d builder nor meshmixer don't report any problems with the mesh. I've created a simpler version of the same object, with only text embossed and that one sliced ok, so it's something about complexity of single layer slicing.

Embossing is for 0.2 z layer height. Not sure how 0.4 would work.

bubnikv, 2021-02-09:
The culprit is the function

```
Polygons BridgeDetector::coverage(double angle) const
```

calculating the area bridged. The implementation is very slow for complex bridging regions. This is the "experimenter's Slic3r" heritage, which needs to be polished.

bubnikv, 2021-02-09:
Fixed with 2e55898.
Funny thing, that extremely slow function produced result, that was never used in production code, only in unit tests. I suppressed the function in production code.

closed 2021-02-09 by bubnikv

foreachthing, 2021-02-11:
Thanks @bubnikv !
It's much faster now ... I wonder if slicing, in general, just became faster?!

bubnikv, 2021-02-11:
> It's much faster now ... I wonder if slicing, in general, just became faster?!

Or just a placebo effect?

Also the fix for #5771 may have shown for you.

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta.