prusa3d/PrusaSlicer #5975: M104 S1500

Bombero-112, 2021-02-05:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Bombero-112, 2021-02-05:
Hallo, ich habe festgestellt das nach dem Slicen einer Zeichnung der Drucker plötzlich auf 1500 grad Temperatur eingestellt wird.

In der Zeile ist auch M104 S1500 vorhanden.

Finde aber die Ursache nicht.

Habe mit Prusa die Datei erstellt. Beim Cura Prog. tritt der Fehler nicht auf.

[M104 S1500](https://user-images.githubusercontent.com/78619812/107085339-a6a03680-67f8-11eb-8d30-71e126978e95.jpg)

Hat hier jemand eine Idee woran das liegen kann.

mfg.

rtyr, 2021-02-05:
You have likely 1500C set in your settings. Please provide your 3MF project file.

rtyr, 2021-02-10:
Google translator:
Sie haben wahrscheinlich 1500C in Ihren Einstellungen eingestellt. Bitte geben Sie Ihre 3MF-Projektdatei an.

rtyr, 2021-03-04:
No response. Closing.

closed 2021-03-04 by rtyr