prusa3d/PrusaSlicer #5976: Empty layers detected (when there is really none rendered), output not printable; possible non-Manifold bugs in Prusa Slicer 2.3

gengimli, 2021-02-05:
### Version
2.3.0

### Operating system type + version
Linux Manjaro KDE 5.20.5 Kernel 5.4.89-1 64 bit

### 3D printer brand / version + firmware version (if known)
Prusa MK3s firmware PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352

### Behavior
Prusa Slicer reports upon slicing
     Warning:
          Empty layers detected, the output would not be printable.
          Object name: xxx
          Print z: 24.35, 22.10
   
[image]

Steps needed to reproduce the problem
     1. Open or import the following file in Prusa slicer 

     2. Click "Slice Now" 
Expected results
     1. for other similar models, the problem does not occur
     2. open the following file

[image]

#### Project File (.3MF) where problem occurs
1. This is the problem file
[Luds 210204 3 round hole.zip](https://github.com/prusa3d/PrusaSlicer/files/5935104/Luds.210204.3.round.hole.zip)

2. This is a working file
[Es 210204 3 round hole.zip](https://github.com/prusa3d/PrusaSlicer/files/5935112/Es.210204.3.round.hole.zip)

I realize replies to this have been for us to check our models and files in meshmixer or netfabb. But do we really have to? Especially when modelling from Blender and even Prusa Slicer shows the models to be intact?

I am also aware that no such problem with Cura occurs.

It seems most of the response of the development team has been to point the issue back to us and have us double check our models and files. Believe me we have. In my case, I have compared the above models in Blender and found them to similarly have similar structures in terms of vertices and meshes.

Why must Prusa Slicer show gaps, empty spaces, and empty layers when there is none on rendered images?

Is there something we need to disable or enable?

I have tried some other suggestions in similar posts including functions as "Detect thin walls", etc. to no avail. I have experimented with infills, and settings. Strangely enough the 0.15 speed and quality print settings seems to suffer less (imagined) errors than other settings. Still its a few errors that render slicing useless. 

I have read that this has something to do with non-manifolds. That is a function I hope Prusa Slicer should simplify or automate as a slicer. We can't be bothered by having to fix meshes especially when Blender, Cura, and Meshmixer certainly finds no problem with in rendering. There seems to be a bug in the current Prusa Slicer configuration that makes it "imagine" or "create" gaps, empty layers, and spaces where there should be none.

In addition, checking the models in Blender shows not much difference in terms of non-manifold errors.

Allow us to 3D print as render as is please Prusa Slicer team. WYSIWYG shouldn't be laborious on our part.

This bug seems to have recurred now and for the past year without being addressed fully.

Will greatly appreciate your attention and solution.

Cubdriver406, 2021-02-06:
I also get the error "Empty layers detected" on a Mac and Slicer version 2.3.0. I solved the problem by reverting back to Slicer 2.9.x and the same model printed without any "Empty Layers". Appears to be a bug in 2.3.0.

gengimli, 2021-02-06:
> I also get the error "Empty layers detected" on a Mac and Slicer version 2.3.0. I solved the problem by reverting back to Slicer 2.9.x and the same model printed without any "Empty Layers". Appears to be a bug in 2.3.0.

Thanks. I'll try your solution. I've been thinking about reverting back but I'm wary that my Prusa Mk3s will demand a firmware rollback.

reklawykSekuL, 2021-02-06:
@gengimli
analyzed your stl file with freeCAD 0.19 and there are a lot of errors (see. screencopy).  So this seems NOT to be a prusa Slicer 2.3.0 bug.

[image]

rtyr, 2021-02-08:
While we may improve repair capabilities of the PrusaSlicer in the future, it doesn't change the fact, that your model is broken (as notified by PrusaSlicer).

Some examples:

**PrusaSlicer:**
[image]

**Simplify3D:**
[image]

**Cura:**
[image]

**Blender** you mentioned:
[image]

gengimli, 2021-02-10:
> @gengimli
> analyzed your stl file with freeCAD 0.19 and there are a lot of errors (see. screencopy). So this seems NOT to be a prusa Slicer 2.3.0 bug.
> 
> [image]
> 
> after a try to repair the errors found there is still an message that the mesh is not solid (second screenshot shows the postion where that obviously occures). It's closed to the position where in prusa slicer the "empty layer" is detected.
> 
> [image]
> 
> Imported again in prusa slicer and repair with the netfab-repair option the part is slicable, but
> you should inspect the slices by dragging the slider to see everything is sliced as you intended.
> 
> [Luds 210204 3 round hole_repaired_Netfab.zip](https://github.com/prusa3d/PrusaSlicer/files/5936727/Luds.210204.3.round.hole_repaired_Netfab.zip)
> 
> good luck
> for the printing
> & best regards from south Hessen
> to the netherlands
> Luke

Thank you Luke;

Took me a few days to not only check for errors but also to print a number of models without the errors.

You are right. My model was not checked for errors and I did so using blender's 3D print toolbox. In it I was able to make manifold those which weren't manilfolds.

I can't take you enough by your clear instructions showing error checking albeit in another program.

all the best,
Emil (gengimli)

gengimli, 2021-02-10:
> While we may improve repair capabilities of the PrusaSlicer in the future, it doesn't change the fact, that your model is broken (as notified by PrusaSlicer).
> 
> Some examples:
> 
> **PrusaSlicer:**
> [image]
> 
> **Simplify3D:**
> [image]
> 
> **Cura:**
> [image]
> 
> **Blender** you mentioned:
> [image]

Thank you rtyr,

Thank you for considering to include some repair facility in future versions too.

My model indeed had a lot of errors. I used an old add-on by tissue guru Alessandro Zomparelli and the errors seem to come innate with the add-on but I'm grateful.

I will post however on another topic a possible request {more like a dream feature} that would allow Prusa Slicer to more accurately depict the possible result of the print given the print and filament settings. I know this is near impossible.

The fact is its still an trial and error when printing as no matter how much corrected the errors are, there are varying results when printing. Some model defects like blobs, strings are still difficult to eliminate despite the errors.

I am grateful however for the error checking inherent in Prusa Slicer.

cheers,
gengimli

gengimli, 2021-02-10:
> I also get the error "Empty layers detected" on a Mac and Slicer version 2.3.0. I solved the problem by reverting back to Slicer 2.9.x and the same model printed without any "Empty Layers". Appears to be a bug in 2.3.0.

I did revert back to Blender LTS over the latest Blender 2.9x more because the add-on I use seems to report and not produce a model citing some python error.

Thanks for your suggestion cubby, at the very least it made me think out of the box and following your logic, I did find a solution that I can apply or keep in my bag of tricks and workaround, not just now but in the future.

cheers,
gengimli

closed 2021-02-21 by gengimli