prusa3d/PrusaSlicer #5977: Freeze on startup application

SPARANGIV, 2021-02-06:
### Version
_Version of PrusaSlicer used_ 

-  2.3.0

_Use `About->About PrusaSlicer` for release versions_ 
 Last  release, could not see it because of the freeze

### Operating system type + version
_What OS are you using, and state any version #s_

- Microsoft Windows 10
- 10.0.19042 N/A Build 19042

### Behavior
On opening the PrusaSlicer the application it self freezes and comse to a hold
* _Steps needed to reproduce the problem_
  * Attempting to opening the program, Already attmepteed to do a reinstallation. Did not help.
* _Expected Results_
  *  The app to open
* _Actual Results_
  * The app freezes on startup
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]
_Is this a new feature request?_
This ain't a request, it seems to be a bug

fe60, 2021-02-12:
The same setup (PrusaSlicer + Operating System) works for me.
There are two things I would try facing this issue:
1. Rename the folder containing the configuration details of PrusaSlicer (%USERPROFILE%\AppData\Roaming\PrusaSlicer) [1] to start PrusaSlicer without an existing configuration.
2. Try out the extracted zip package instead of the installer.

[1] https://help.prusa3d.com/en/article/faq-prusaslicer_1789
[2] https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.0

SPARANGIV, 2021-02-13:
@fe60 I was looking for this folder, rename and reconfiguring did the trick. I think something went corrupt in there.

> 1. Rename the folder containing the configuration details of PrusaSlicer (%USERPROFILE%\AppData\Roaming\PrusaSlicer) [1] to start PrusaSlicer without an existing configuration.

[image]

Other tests:
- Second run, without any problem ✔
- Run from Meshmixer, also working again. ✔

[image]

Thanks a lot

SPARANGIV, 2021-02-13:
Hey I'm back again, It has something to do with moving to a other monitor.

Displaysetup triple 1920x1080
[image]

I will look in the `%USERPROFILE%\AppData\Roaming\PrusaSlicer` files if I can find somthing off

Greatings

SPARTANGIV, 2021-02-13:
So this is what I did, in the '%USERPROFILE%\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` I removed the folowing line `window_mainframe = -3848; -8; 1936; 1056; 1`  this resolved the problem. 

Does someone see somthing perticaler off in this part? `window_mainframe = -3848; -8; 1936; 1056; 1` I do understand it are the pixels, I don't understand why it would result in a freeze because it seem to be legit values to me

Greetings

bubnikv, 2021-09-25:
May be similar to https://github.com/prusa3d/PrusaSlicer/issues/2939

SPARTANGIV, 2021-09-26:
@bubnikv , thx for the command. But I do no longer have a problema afther the cange is made to the ini file. #

> So this is what I did, in the `%USERPROFILE%\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` I removed the folowing line `window_mainframe = -3848; -8; 1936; 1056; 1` this resolved the problem.
> 
> Does someone see somthing perticaler off in this part? `window_mainframe = -3848; -8; 1936; 1056; 1` I do understand it are the pixels, I don't understand why it would result in a freeze because it seem to be legit values to me.
> 
> The values placed back by the slice are `window_mainframe = -8; -8; 1872; 1096; 1` this seems to work fine.
> 
> Greetings

Williamson-Matt, 2021-11-03:
am able to reproduce this by full-screening PrusaSlicer on 2nd (of three) monitor, closing, and reopening. this resulted in the program freezing. Issue is fixed by deleting `window_mainframe` property, and brought back by full screening and closing the program while it is full screened on the second monitor. 

my ini setting: `window_mainframe = -1928; -8; 1936; 1176; 1`

monitor arrangement is `2 | 1 | 3`

SPARTANGIV, 2021-11-03:
> my ini setting: `window_mainframe = -1928; -8; 1936; 1176; 1`

Could you try this valu `window_mainframe = -8; -8; 1936; 1176; 1`

Greetings

bubnikv, 2021-12-08:
We may do something like Cura guys do: Let the user disable restoring window position on start.

[image]

However one would have to edit PrusaSlicer.ini manually to disable restoring window positions if it causes issues.

lukasmatena, 2022-04-05:
@GvBeers Do you still have the issue with PrusaSlicer 2.4.1? We implemented something similar to Cura (see @bubnikv 's post above mine). You can disable the position restoring in Preferences, it is apparently what causes the crash under some scenarios. It should disable automatically in case the crash occurs and notify the user on the next start that it happened, so the manual editing on PrusaSlicer.ini to get PrusaSlicer start again should not be needed.

GvBeers, 2022-07-28:
> @GvBeers Do you still have the issue with PrusaSlicer 2.4.1? We implemented something similar to Cura (see @bubnikv 's post above mine). You can disable the position restoring in Preferences, it is [...]

@lukasmatena I haven't opent Prusa for a wille, also becouse of the edit the crash did not happen again, so I could't tell. I will let you know if I update the Slicere, becouse I wanted pick up the printing again.

lukasmatena, 2023-02-22:
Hopefully fixed. Closing.

closed 2023-02-22 by lukasmatena