prusa3d/PrusaSlicer #5979: Prusa slicer segfaults when using the --split option from the command line

NuclearLighthouseStudios, 2021-02-06:
### Version
2.3.0, though this same bug has existed in previous versions

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Prusa i3MK3S

### Behavior
When slicing an object from the command line using the `--split` option will result in a segmentation fault.  
This seems to happen regardless of the mesh, wether is actually has multiple parts that can be split, and the slicing settings.

#### Steps needed to reproduce the problem
Run `prusa-slicer` on any stl or 3mf using the `--split` option, for example:  
`prusa-slicer -s --split boxes.3mf`

#### Expected Results
The mesh gets split into multiple parts and sliced.

#### Actual Results
Prusa-slicer segfaults.

#### Example

```
$ prusa-slicer -s boxes.3mf
10 => Processing triangulated mesh
20 => Generating perimeters
70 => Infilling layers
30 => Preparing infill
88 => Generating skirt
90 => Exporting G-code to boxes_0.2mm_PLA_MK3S_{print_time}.gcode
Slicing result exported to boxes_0.2mm_PLA_MK3S_5h35m.gcode
$ prusa-slicer -s --split boxes.3mf
Segmentation fault (core dumped)
$ 
```

#### Example files
Both of these files reproduce the error when prusa-slicer is run on them using the `--split` option.
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/5936764/files.zip)

combolek, 2021-02-08:
@NuclearLighthouseStudios Can you try PR #5991 ?

NuclearLighthouseStudios, 2021-02-08:
Confirmed working!

combolek, 2021-02-08:
Cool, thanks for testing. I think Lukas is working on a better fix.

lukasmatena, 2021-02-10:
Should be fixed with 22b2ccc for 2.3.1. Thanks @combolek for the fix provided in #5991.
Closing.

closed 2021-02-10 by lukasmatena