prusa3d/PrusaSlicer #5980: [Feature Request] Show estimated length of filament used in G-code Viewer

popecix, 2021-02-06:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 and OSX Big Sur

### 3D printer brand / version + firmware version (if known)
Prusa MK3s 3.9.3
CR-10 Clone Marlin

### Behavior
_Is this a new feature request?_
yes

Description of feature request:
Since its release, I've been using the Prusa G-code viewer almost exclusively (love the speed and lightweight footprint, btw!) except for when I need to check if I have enough of a particular filament for a specific print, i.e. when I want to use up the rest of a spool. In which case I have to either load it back into the full program and reslice (longer process), use Pronterface (longer load time), or go to a website (gcode.ws) to see how many meters are to be used.

Since it shows up in Pronterface and also a website can determine the expected length, I'm presuming that it's calculated from the gcode itself. Would it be possible to simply also display it in the "Feature Type" view somewhere? Similar to how the estimate print time is shown?

I realize it's a QoL thing, but know I would find it very useful and hope some others would as well. Scanned through existing issues/FR and didn't see this specific thing, so tossing it out there for consideration.

Thanks for all your awesome work!

LMF5000, 2021-02-08:
I'm just a regular user here but I thought I'd chime in to say I second this - it's a really useful idea.

P.S. As a workaround for now you can go to gcodeanalyser.com to get this data (and much more)

foreachthing, 2021-02-08:
I just use a simple spreadsheet for that... Enter volume from slicer => length in m, and revolutions on spool (easy to count and accurate enough).

popecix, 2021-02-08:
I should also add I have found it is possible to include the "used_filament" variable in the Output Filename Format of print settings. Though it's shown as the full precision (6 characters, i.e.: 51.6889) and it's not something I'd want to include globally on all gcode files.

rtyr, 2021-08-12:
There is now more info regarding material used. It will be included in the upcoming PrusaSlicer 2.4.0-alpha1. But I guess you would like to see total amount.

[image]

popecix, 2021-08-13:
Yeah, I often slice models when I'm away from home and get a queue of them ready to print for when I get back. The problem I often run into is not knowing if I have enough of a given filament left to actually do the prints. So rather than having to basically re-slice the project again to verify how much is needed, it would be great to also see the total mass (grams) or the length (meters) of filament a print requires in the G-Code Viewer the same as is shown in the slicer after the gcode has been generated:

[image]

Hologos, 2022-01-31:
The information is included in the gcode as a comment. I don't know why total length and weight is not shown. 🤷‍♂️

lukasmatena, 2022-02-01:
The total filament length is available in 2.4.0 GCode Viewer. You just have to switch View to "Tool".

Hologos, 2022-02-01:
> The total filament length is available in 2.4.0 GCode Viewer. You just have to switch View to "Tool".

Didn't notice, thank you. Still I think it would be better to see the total in Feature view since there are all parts of the total.

dartrax, 2022-07-24:
The idea of totals in the legend is elaborated more in #8208.

drf5n, 2023-08-31:
The sliced total weight and length shows up on the Plater tab report at the bottom left:

[image]

I often divide the length in meters by the pi and a 0.1m spool diameter to eyeball if there are enough turns on the spool to complete the print:

    7.65m/pi/0.1m = 22 turns

closed 2026-09-29 by prusaslicer-bot[bot]