prusa3d/PrusaSlicer #5982: Fine details are lost after slicing

mluckau, 2021-02-06:
### Version
2.3.0

### Operating system type + version
Linux 5.10.7-3-MANJARO

### 3D printer brand / version + firmware version (if known)
Tronxy X5SA-Pro Marlin bugfix-2.0.x (build 06.02.2021)

### Behavior
Fine details are lost after slicing.
As you can see on the screenshot, the numbers on the box are no longer completely present after slicing.
It can't be due to the model, because it works perfectly under cura. 
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
[Nozzle_Box.zip](https://github.com/prusa3d/PrusaSlicer/files/5937583/Nozzle_Box.zip)

foreachthing, 2021-02-07:
It's "Detect thin walls" ("Dünne Wände erkennen"). Disable it in "Druckeinstellungen" -> "Qualität".
Dann sollte es so herauskommen wie in Cura.
[image]

mluckau, 2021-02-08:
Thank you, that actually fixed it. I didn't think it was this option. Again something learned. Thank you!

closed 2021-02-08 by mluckau