prusa3d/PrusaSlicer #5983: Possible bug - External Perimeter speed wrong

n9jcv, 2021-02-07:
### Version
Version 2.3.0+win64
Windows 10

Ender 3 clone (Voxelab Aquila) Marlin 2.0.6 - Vendor firmware

Printing an object and the external perimeter line is basically not there.  I investigate the gcode, and it seems to me that PrusaSlicer is incorrectly setting the speed with this command;
G1 F2400.000

The object is attached.  First two layers print fine.  Problem is on third layer.  Here is part of the suspect GCODE from the 3rd layer;  My settings - perimeters 50mm/s and external perimeters 40mm/s    so my external perimeter should be slower than the internal perimeters, but when I watch the printer, it go several times faster on the external perimeter than the internal perimeters, and the filament does not flow out fast enough, leaving the external perimeter basically empty.  I will attach the gcode file, and the stl.  Please print the first 3 layers of this very small object for yourself and you should see the same issue.  This issue does not happen on other layers in this object, only this layer, and I do not know why, but it is repeatable.  The object was larger and it had the same results.  I cut the 4 layers from a much larger object to try and isolate the issue.

I have also attached an image of the toolpath on the 3rd layer.  The yellow 4 inner perimeters works fine.  The orang external perimeter is where I am seeing the problem.  Please let me know if I can assist in any other way.  Thank You for all your work on this Prusa Slicer!!!!

LAYER_CHANGE
;Z:0.84
;HEIGHT:0.28
;WIPE_START
G1 F4320.000
G1 X118.192 Y99.765 E27.54728
G1 X118.192 Y100.253 E27.35410
G1 X119.072 Y99.373 E26.86143
G1 X119.134 Y99.373 E26.83701
G1 X119.134 Y99.799 E26.66825
G1 X118.192 Y100.741 E26.14104
G1 X118.192 Y101.229 E25.94787
G1 X119.134 Y100.287 E25.42066
G1 X119.134 Y100.775 E25.22748
G1 X118.192 Y101.717 E24.70027
G1 X118.192 Y102.205 E24.50710
G1 X118.835 Y101.562 E24.14739
;WIPE_END
G1 E23.95239 F1800.000
G92 E0
G1 Z0.840 F5400.000
G1 X118.663 Y113.289
G1 E3.90000 F1800.000
;TYPE:Perimeter
;WIDTH:0.45
G1 F900.000
G1 X117.948 Y113.289 E3.93247
G1 X117.948 Y99.128 E4.57523
G1 X119.378 Y99.128 E4.64017
G1 X119.378 Y113.289 E5.28293
G1 X118.723 Y113.289 E5.31267
G1 X118.663 Y113.679 F5400.000
G1 F900.000
G1 X117.558 Y113.679 E5.36284
G1 X117.558 Y98.738 E6.04100
G1 X119.768 Y98.738 E6.14133
G1 X119.768 Y113.679 E6.81949
G1 X118.723 Y113.679 E6.86693
G1 X118.663 Y114.069 F5400.000
G1 F900.000
G1 X117.168 Y114.069 E6.93480
G1 X117.168 Y98.348 E7.64835
G1 X120.158 Y98.348 E7.78408
G1 X120.158 Y114.069 E8.49763
G1 X118.723 Y114.069 E8.56277
G1 X118.663 Y114.459 F5400.000
G1 F900.000
G1 X116.778 Y114.459 E8.64834
G1 X116.778 Y97.959 E9.39728
G1 X120.548 Y97.959 E9.56841
G1 X120.548 Y114.459 E10.31736
G1 X118.723 Y114.459 E10.40020
G1 X118.663 Y114.849 F5400.000
;TYPE:External perimeter
G1 F2400.000
G1 X116.388 Y114.849 E10.50346
G1 X116.388 Y97.569 E11.28780
G1 X120.938 Y97.569 E11.49433
G1 X120.938 Y114.849 E12.27867
G1 X118.723 Y114.849 E12.37921
G1 X118.326 Y114.634 F5400.000
G1 E8.47921 F1800.000
G92 E0
G1 X119.287 Y99.926 F5400.000
G1 E3.90000 F1800.000

[testgcode.txt](https://github.com/prusa3d/PrusaSlicer/files/5938471/testgcode.txt)

[test.zip](https://github.com/prusa3d/PrusaSlicer/files/5938473/test.zip)
[image]

rtyr, 2021-02-07:
The inner perimeters are likely printed with "small perimeter speed", because of its small radius.

rtyr, 2021-02-09:
Can we close the issue?

n9jcv, 2021-02-09:
Yes please close. It is true the perimeter was printed at small perimeter setting

closed 2021-02-09 by rtyr