prusa3d/PrusaSlicer #5984: Can not connect to Octoprint

markdesouza, 2021-02-07:
### Version
2.3.0+

### Operating system type + version
MAC OS 11.2

### 3D printer brand / version + firmware version (if known)
N/A (but its a Prusa MK3s)

### Behavior
Can not connect to my octoprint server. Entering: octoprint.home in the hostname/ip/url field

Prusaslicer returns: 
Could not connect to OctoPrint: SSL peer certificate or SSH remote key was not OK:
SSL: certificate verification failed (result: 5)
[Error 51]
Note: OctoPrint version at least 1.1.0 is required.

I have a private CA and it has been imported into my keychain and the certificate has been flagged as trusted.

Chrome / Firefox / Safari can all browse to https://octoprint.home (and state the certificate is valid)
Running from the terminal: curl https://octoprint.home (returns the html of the octoprint login page)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
N/A

ksvladimir, 2021-03-17:
Hi, I have exactly the same issue. SSL works in the browser and in terminal using curl, but not i prusa. I run from cmdline with `--loglevel 5`, here are the logs:

```
[2021-03-17 11:14:41.710169] [0x00000002040dee00] [info]    OctoPrint: Get version at: https://octopi.local/api/version
*   Trying fe80::7a44:5ec:fac:fe9a...
* TCP_NODELAY set
* Connected to octopi.local (fe80::7a44:5ec:fac:fe9a) port 443 (#0)
* SSL: certificate verification failed (result: 5)
* Closing connection 0
[2021-03-17 11:14:41.867018] [0x00000002040dee00] [error]   OctoPrint: Error getting version: SSL peer certificate or SSH remote key was not OK:
SSL: certificate verification failed (result: 5)
[Error 51], HTTP 0, body: ``
[2021-03-17 11:15:20.689141] [0x00000002040dee00] [debug]   message handler shutdown().
```

I believe this might also be related to this discussion: https://forum.prusaprinters.org/forum/prusaslicer/octoprint-https-validation-on-mac-with-2-3-0/ .

jlyonsmith, 2022-05-29:
I just hit this issue as well. Looks like the fix is for the developers to switch to OpenSSL on MacOS.  In the meantime, I fixed it by disabling the redirect from HTTP to HTTPS (as described [in this article](https://community.octoprint.org/t/setup-a-new-self-signed-ssl-certificate-on-octoprint-enable-https/30256)) and PrusaSlicer defaults to HTTP.

bubnikv, 2022-05-30:
We will switch to libcurl provided by Cuppertino as part of the operating system. libcurl provided by Apple is compiled against OpenSSL. Apple does not eat its own dog food (sic!) and their documentation is ** of five.

The change should be released with PrusaSlicer 2.5.0 relatively soon.

bubnikv, 2022-06-29:
Implemented in PrusaSlicer 2.5.0-alpha2. Closing.

closed 2022-06-29 by bubnikv