prusa3d/PrusaSlicer #5989: Slicing issue, and bed1_temperature not found

joncl, 2021-02-07:
### Version
2.3.0 win64

### Operating system type + version
Windows 10 64-bit

### 3D printer brand / version + firmware version (if known)
Tevo Little Monster
Board: Duet WiFi 1.02 or later
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 2.05 (2019-12-13b1)
Duet WiFi Server Version: 1.23

### Behavior
Using Simplify3D for a long time, no main issue. But PrusaSlicer looks better! First time setting up PrusaSlicer. Manually entered all working settings from Simplify3D. PrusaSlicer is not slicing correctly for some reason.

There's also an error encountered when slicing:
Parsing error at line 2: Variable does not exist
M140 S[bed1_temperature] T0

I have the bed temperature set, so I'm not sure what's happening there.

Thanks for taking a look!

#### Project File (.3MF) where problem occurs

Exported files
[fan_shroud_v340.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5939703/fan_shroud_v340.3mf.zip)
[fan_shroud_v340.amf.zip](https://github.com/prusa3d/PrusaSlicer/files/5939699/fan_shroud_v340.amf.zip)
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/5939700/PrusaSlicer_config_bundle.zip)

Screen shots:
[image]
[image]
[image]

rtyr, 2021-02-07:
> Parsing error at line 2: Variable does not exist
> M140 S[bed1_temperature] T0

It is Simplify3D specific placeholder, it cannot be used in PrusaSlicer. 

Use this for extruder/bed temperature:
[first_layer_temperature]
[first_layer_bed_temperature]

The weird looking slicing is caused by your resolution setting, change it back to 0.

[image]

joncl, 2021-02-07:
On bed temp, that's interesting because I just imported a STL file into
Prusa. No idea how it got the Simplify3D variable in there. I'll check it
out. Thank you!

lukasmatena, 2021-02-08:
This can be closed I guess.

closed 2021-02-08 by lukasmatena

joncl, 2021-02-11:
Hello, thank you very much. Those tips resolved my issue (Resoluton=0 and
temp variables). I'm running into another issue on this same model. The
extrusion width is 1/2 or less of what it should be. Should I post
separately? My nozzle width is .4mm and extrusion widths are all 100% (with
default at .48mm). These are the same settings as in Simplify3D. Thanks!

Screen grabs below with quick notes...

In PrusaSlicer I get:

[image: image.png]

In Simplify3D I get (working print):

[image: image.png]

rtyr, 2021-02-11:
PrusaSlicer behaves differently in this regard. It is calculated over layer height if you enter percentage. I can understand, that it can be confusing. It is better to use exact values (for example 0.48).

[image]

joncl, 2021-02-12:
Ok, yes. I didn't understand the part: "if expressed as a percentage, will
be computed over layer height", because I don't understand how extrusion
width is related to layer height. But anyway, I set all values to actual
widths, and got the expected slice. It's now printing the part! Actually,
it's really impressive! The printing process seems to be much smoother vs.
S3D. Definitely, the process is much more refined. I also like the support
structures much better. Plus, the gcode file is 1/2 the size! (17MB vs
8MB). Really cool!! PrusaSlicer is the way to go now! Thanks very much for
your help getting this going!