prusa3d/PrusaSlicer #5993: infill percentage not changing

cabbey, 2021-02-08:
### Version
aa4c018cd5857814ed9ac0cd93d526f3f172d4ab is the commit that introduced this issue. previous commit worked fine.

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Prusa mk2.5s MMU2S

### Behavior
infill percentage settings no longer impact the actual infill. 10%, 50%, 90% all produce the same result. Note this is not just a rendering issue, the output gcode is the same.
[image]
[image]

#### Project File (.3MF) where problem occurs
[bug-report.zip](https://github.com/prusa3d/PrusaSlicer/files/5940932/bug-report.zip)

cabbey, 2021-02-08:
tagging @bubnikv

jmd73, 2021-02-08:
It works for me on Windows 10 with 2.3.0 release version.

After writing a value:
But I have to click somewhere outside the infill percentage combobox or move the mouse pointer over the 3d-scene (that seems to make the combobox lose focus.)

Pressing the enter key does nothing after writing a value.

Choosing a value with the combobox drop-down list:
Triggers a re-slicing with the new value.

bubnikv, 2021-02-08:
Fixed.

closed 2021-02-08 by bubnikv

cabbey, 2021-02-09:
> It works for me on Windows 10 with 2.3.0 release version.

This bug was raised for a change in a source commit dated well after 2.3.0 was released. Whatever you were attempting to discus here @jmd73 is unrelated.

> Fixed.

:facepalm: I looked at that code and thought "yeah, 0 to 1... makes perfect sense."
Confirmed the fix, thanks!