prusa3d/PrusaSlicer #5995: Max Volumetric Speed creates gcode with higher volumetric speed than is set as maximum

honuvo, 2021-02-08:
### Version
PrusaSlicer Version 2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro unmodified with stock firmware 1.1.6.2

### Behavior
Example:

Settings:
Max Volumetric Speed: 4mm3/sec
All Speeds to 60mm/sec
Layer height 0.28mm
Line Width 0.45mm
Nozzle: 0.4mm
Minimum Layer Time 1sec (to not interfere)

Manually calculating, the maximum speed in this scenario would be 4 / 0,45 / 0,28 = 31,75mm/s
When checking the gcode (via i.e. gcode.ws) most of the extrusions happen at 36,64mm/s what would calculate to a volumetric speed of 4,62mm3/s which is over 15% more.

Is this a new feature request?
No

#### Project File (.3MF) where problem occurs
[20mm_calibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/5944995/20mm_calibration_cube.zip)

enricoturri1966, 2021-02-09:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5765 ?

honuvo, 2021-02-09:
> Duplicate of #5765 ?

Hi, i've read that issue before posting, but in that case it seems that only small portions or one single extrusion seems to go over the limit what let's the scale shoot up. In my case the whole print is too fast/too much extrusion.

lukasmatena, 2021-12-06:
@honuvo I suggest to look at https://manual.slic3r.org/advanced/flow-math
Your assumption of rectangular extrusion is not what PrusaSlicer uses. It assumes that the cross section has rounded sides, and the area difference is exactly 15 % when I plug in your numbers.

Closing.

closed 2021-12-06 by lukasmatena