prusa3d/PrusaSlicer #5996: Anycubic i3 Mega S printsettings start G-Code

AndiP1234, 2021-02-08:
### Version
_V2.3.0+win64bit_

### Operating system type + version
win10 64bit

### 3D printer brand / version + firmware version (if known)
Anycubic i3 Mega S Marlin 1.9

### Behavior
at the Anycubic ver 0.0.6 (i3 Mega S) settings is a bug in the start g-code of the print settings. The z position for the intro line is set fix to 0.
If some needs to use the z_offset to fix the wrong adjustment of the limitswitch and the nozzle is to low, while the intro line the nozzle can destroy the bed.

Bug:
`G1 X60 Z0 E9 ; intro line`

Solution:
`G1 X60 Z{z_offset} E9 ; intro line`

`G90 ; use absolute coordinates
M83 ; extruder relative mode
M221 S100 ; set Flowrate to default
M204 S[machine_max_acceleration_extruding] T[machine_max_acceleration_retracting]
;M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
G28 ; home all
G1 Y0 Z1 F100 ; move print head up
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G92 E0
G1 E38 F1000; deretract filament
G92 E0
G1 X60 Z{z_offset} E9 ; intro line
G1 X100 E12.5 ; intro line
G92 E0`

rtyr, 2021-02-08:
It is certainly not ok, thanks. I will change it (but not with the suggested solution). 

I am not sure, why @igami uses such start g-code.

rtyr, 2021-02-08:
Fixed. Closing.

Igami, 2021-02-09:
> I am not sure, why @Igami uses such start g-code.

At the end of the G-code, the filament is retracted so that it cools outside the hotend and can be easily changed without preheating.
But when you start a print and the filament is still loaded, there is a huge blob at the nozzle. So I lifted the nozzle to 1mm and then go down to 0mm with the intro line.

But as you suggest, it shouldn't be 0mm. Maybe {z_offset + 0.2}.

rtyr, 2021-02-09:
@Igami 
Please test whether current hotfix works. I placed Z movement in front of the purge line, which may be not optimal for your strategy. I don't like the retract/deretract thing to be honest, maybe we should remove it. It is quite unusual and not safe, since the initial state of the filament is not known.

I think the z_offset shouldn't be used on most of the printers. It should be fixed on the printer side.