prusa3d/PrusaSlicer #6002: sequential printing not being honoured or cutting feture required please

pcsentinel, 2021-02-09:
### Version
2.3.0

I often print using lily pads one in each corner of a rectangular print. Ideally I either want to print the first layer of the body of the main object first, and then the which are only 1 layer high. OR the facility to automatically cut the lily pads by the main body.

I have tried ordering the pads in the object list, but it doesn't change the order in slicing. So on the attached image, 2 pads get printed then the first layer of the main body and then the other 2 pads.

Thanks, your doing a great job BTW.

[image]

rtyr, 2021-02-09:
Right click on the main model and select "Add part". The pads will be merged with the model.

[image]

pcsentinel, 2021-02-09:
Hi, sorry but that means that the perimeter of the main part is not printed where it coincides with the part, visually this is noticeable.

[rc]<https://user-images.githubusercontent.com/36745189/107413791-72d35280-6b11-11eb-9347-894b842975ac.jpg>

rtyr, 2021-02-09:
I think it will be better than printing over already printed part (like on your screenshot). Some sort of mouse/brim ears feature is on our list of possible improvements (there are also already some issues opened regarding that).

rtyr, 2021-02-09:
You should be able to achieve that, if you set different parameters for the pads. 

1 perimeter for the pads in this case:
[image]

pcsentinel, 2021-02-10:
Hi, thanks for your help so far, but I can't get that to work.

Firstly, I have found a bug(ette) If a part is rotated and then you use the add function then the added part is not placed on the workplane, the part below has been rotated -90 degs and then right click on part, add lily pad, ends up with the pad in mid air.

[cid:d8fe2bd7-767c-48a5-8abc-7616810442e5]

If I export the part post rotation, and then open it, the lily pad is added on the bed. however, I set the perimeters just on the pad, and then perimeters are right, but they are still crossing the main part, which I don't see in your image.

[cid:1f2788c4-eb98-4c20-9ee1-d791beb5e57c]

thanks

Paul

[11]<https://user-images.githubusercontent.com/36745189/107433609-da959780-6b29-11eb-98bf-824fac321a76.jpg>

rtyr, 2021-02-10:
Please provide your 3MF project.

pcsentinel, 2021-02-10:
Hi, 2 files attached, the first is the rotated part

The second is the rotated part, exported as an stl then imported to a new project and the pads added.

I have also attached the pad stl

thanks

Paul

rtyr, 2021-02-10:
You have to attach it using github web interface.

pcsentinel, 2021-02-10:
Sorry

[Pads.zip](https://github.com/prusa3d/PrusaSlicer/files/5958199/Pads.zip)

pcsentinel, 2021-02-11:
Hi, just to let you know that I have sort of achieved a solution, its not as pretty as the one you sent the picture of. But If I add main part, then 4 lily pads as separate objects, then select all 4 pads and merge them, so that I essentially have 2 objects. I then move the main object to below the pads in the object list. The result is  a slice that does the first layer of the main object first then the pads. The remaining issue of course is that where the pads overlay the main object both are printed. I still cant achieve  the output from your image though.

closed 2026-09-29 by prusaslicer-bot[bot]