prusa3d/PrusaSlicer #6007: Prusaslicer 2.3 printerwizard

ThomasX1, 2021-02-10:
Prusaslicer 2.3

In the printerwizard is other printers like Artillery disappeared ?
[image]

rtyr, 2021-02-10:
No, Artillery was never there. At least not from us.

rtyr, 2021-02-10:
Vendor list from your screenshot is OK, nothing is missing. There is a new pull request with some Artillery profiles so we may include it in the next PrusaSlicer release (after review).

Closing.

closed 2021-02-10 by rtyr

ThomasX1, 2021-02-10:
Ok, but i dont understand because on facebook i've found this. Is this fake or not reliable?
[image]
[image]
[image]

rtyr, 2021-02-11:
That is custom bundle (manually added, not bundled by us). It is actually made by the same author who created pull request yesterday.