prusa3d/PrusaSlicer #6008: bonjoçur, impossible ajouter filaments (par exemple flex) (version 2.3)

seniorpapou, 2021-02-10:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-02-10:
Please provide some description, what printer are you using etc.

seniorpapou, 2021-02-10:
Bonjour, 
merci d'avoir répondu
sur la première image la partie droite donne des infos de base 

[image]
sur la deuxième a partie droite affiche les propositions de base 
[image]
sur la troisième la partie droite me permet de faire mes choix, 
[image]
Je clique sur "FIN"
sur la quatrième, je ne vois pas les produits ajoutés
[image]
Merci

rtyr, 2021-02-10:
Those filament profiles are compatible only with MMU2 in single mode (not for MM mode). Same issue as https://github.com/prusa3d/PrusaSlicer-settings/issues/97. Closing.

closed 2021-02-10 by rtyr

seniorpapou, 2021-02-10:
Bonjour,
après essais, cela se produit dans le cas mmu2s 
et cela fonctionne dans le cas mmu2s single
Cordialement