prusa3d/PrusaSlicer #6010: wrong Z height when printing multiple objects

PhilippeAB, 2021-02-10:
### Version
2.3.0

### Operating system type + version
MacOS 10.15.7

### 3D printer brand / version + firmware version (if known)
Eryone Thinker SE / Marlin 2.0

### Behavior
When printing multiple items the result isn't the same as printing just one item 
Z layers are crushed and  there is some overextrusion but it does not look linear

[fume_extractor.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5961538/fume_extractor.3mf.zip)

rtyr, 2021-02-11:
I am sorry, but I don't understand what do you mean.

What can I see in your project is that you mostly copied our Prusa MINI profile, including machine limits. I doubt, that those machine limits are valid for your printer. You should modify those values or at least not emit them into the gcode. With your current settings, those machine limits will override your firmware limits. It can lead to various issues, including skipped steps on Z axis.

PhilippeAB, 2021-02-11:
I use the eryone prusaslicer profile. I know it's derivated from Prusa Mini's profile but it work well when there is one object on the bed. The various speed/acceleration settings  should be correct. 
I can retry with prusaslicer 2.2 but I never had such problem printing multiple items at once before
It's also print nicely when slicing the project with Cura (But yes it's not the same profile).
I will try on the prusa mini to check.
I can post picture of the print if you want.

rtyr, 2021-02-11:
If printed objects are smaller in Z dimension and some layers are overextruded then it looks like skipped steps on Z axis. It may be hardware issue bust most likely you have just incorrect values in machine limits (too high speed/acc for z axis).

More objects = more z-hops.

Switch this to "Use for estimate" or to "Ignore", reslice the g-code, restart the printer and try to print it again.
[image]

rtyr, 2021-02-11:
Just as I suspected. Maximum speed for Z axis in stock Eryone firmware is 5mm/s. You are overriding it to 12mm/s in PS settings.

[image]

PhilippeAB, 2021-02-11:
OK I don't know why the profil they provide is wrong.
Bur you are right. I also checked in Cura and there is no Z hop on retract.
Sorry about this

rtyr, 2021-02-11:
No problem.

Closing.

closed 2021-02-11 by rtyr