prusa3d/PrusaSlicer #6012: MK3s + MMU2s. PRINT BEGIN WITH TEMPERATURE OF FILAMENT N°1 ALSO IF YOU DON'T USE

SNDR0, 2021-02-10:
### Version
2.3.0
### Operating system type + version
WINDOWS10 64BIT 20H2
### 3D printer brand / version + firmware version (if known)
ORIGINAL PRUSA MK3s + MMU2s. 
### Behavior
IF YOU HAVE DIFFERENT TYPE OF FILAMENTS CONFIGURED IN PRUSA SLICER (FOR EXAMPLE N°1 PLA, N°2 PETG, N°3 PETG...) AND YOU DON'T USE THE FIRST FILAMENT (PRINT BICOLOR USING N°2 AND N°3), WHEN THE PRINT BEGIN YOU WILL SEE THAT TEMPERATURES ARE FOR PLA... THE PROBLEM IS SOLVED IF YOU CHANGE FIRST FILAMENT TO THE CORRECT TYPE ALSO IF NOT USED.

rtyr, 2021-02-11:
Please provide your 3MF project file, where the problems occurs.

SNDR0, 2021-02-11:
unable to recreate the problem..

closed 2021-02-11 by SNDR0