prusa3d/PrusaSlicer #6014: [Feature Request] Solid Infill Fill Pattern Control

AdamLen, 2021-02-11:
### Version
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Printer: Prusa Research i3 MK3S+
Firmware: 3.9.3_3556
Modifications: Bear Z-Axis

### Behavior
**THIS IS A NEW FEATURE REQUEST**

Ability to control solid infill pattern to match that of first and or top fill pattern for aesthetics of print. I imagine adding this feature wouldn't be extremely difficult to add and I would likely try my hand at it if I had the time and was more comfortable with how PrusaSlicer is programmed.  What I imagine is just having the same dropdown options for solid infill pattern to that of first and top fill patterns while still maintaining fill angle control.

[image]
Figure 1: The Part Being Printed

[image]
Figure 2: First Layer Infill Pattern (Concentric)

[image]
Figure 3: Infill Settings

[image]
Figure 4: Printed As Spiral Vase

[image]
Figure 5: Top Fill Pattern (Concentric)

[image]
Figure 6: Desired Thickness of Solid Infill

[image]
Figure 7: Only Possible Solid Infill Pattern 1

[image]
Figure 8: Only Possible Solid Infill Pattern 2

#### Project File (.3MF) where problem occurs
Project File: [PrusaSlicer_Project_File.zip](https://github.com/prusa3d/PrusaSlicer/files/5962427/PrusaSlicer_Project_File.zip)

#### Closing
Thank you for your time and I'm excited to hear people's opinions if this is doable for PrusaSlicer.

rtyr, 2021-02-11:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5803.

AdamLen, 2021-02-14:
@rtyr I suppose but it wouldn't apply in the case of spiral vase since that is 0% infill so that setting would not carry over to solid infill.

closed 2026-09-29 by prusaslicer-bot[bot]