prusa3d/PrusaSlicer #6015: Bug: Certain STLs crash PrusaSlicer when trying to use Hilbert Curve top/bottom layers

LordMelkor, 2021-02-11:
### Version
2.3.0+win64

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
MK3S latest firmware (also, doesn't matter)

### Behavior
* The STL in the attachment will not slice if attempting to use hilbert curve for the bottom layer.
* Slicing will hang and eventually freeze the program.

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs

[Chrysanthemum.zip](https://github.com/prusa3d/PrusaSlicer/files/5962483/Chrysanthemum.zip)

combolek, 2021-02-11:
This is the same as #5771 and was fixed in 672ed99320ba7e57fa5c1daad3b450ca6320b2f1

bubnikv, 2021-02-11:
duplicate, closing.

closed 2021-02-11 by bubnikv