prusa3d/PrusaSlicer #6020: Travel speed within first layer doesn't honor "First layer speed"

delaosa, 2021-02-11:
### Version
2.3.0

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
FLSUN Q5

### Behavior

"First layer speed" setting is ignored regarding travel speed, it uses the general value instead of been limited to first layer setting. According to the info of this setting, it applies to all moves of the first layer.

Expected: Travel speed in first layer is limited to "Fist layer speed" parameter.

Observed: Travel speed is following the general (rest of leayers) "Travel" parameter.

[image]

rtyr, 2021-02-11:
It does what the tooltip says. Travel is non-print move. Closing.

[image]

closed 2021-02-11 by rtyr

delaosa, 2021-02-12:
My mistake,  it's crystal clear. I already saw some issues asking for more control for the first layer, I'll post it there.

pmarks-net, 2024-02-26:
I implemented this as a post-processing script: https://github.com/prusa3d/PrusaSlicer/issues/1424#issuecomment-1963347733