prusa3d/PrusaSlicer #6023: G-code flavor "No extrusion" not working like description in tooltip suggests

th33xitus, 2021-02-11:
### Version
2.3.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
Firmware: Klipper

### Behavior
The g-code flavor "No extrusion" is described as preventing PrusaSlicer from exporting any extrusion value at all.
This at least is the description from the g-code flavor tooltip.

In reality, PrusaSlicer does export something. To be more specific, the generated g-code will look something like this:
```g-code
G1 X68.387 Y53.898 0.00000
G1 X70.341 Y51.840 0.00000
G1 X72.431 Y49.857 0.00000
G1 X74.587 Y48.014 0.00000
G1 X76.170 Y46.778 0.00000
G1 X78.512 Y45.109 0.00000
G1 X80.910 Y43.583 0.00000
G1 X83.417 Y42.164 0.00000
G1 X85.995 Y40.879 0.00000
G1 X88.596 Y39.747 0.00000
G1 X90.471 Y39.027 0.00000
G1 X93.204 Y38.115 0.00000
G1 X95.938 Y37.358 0.00000
G1 X98.751 Y36.735 0.00000
```
This is the part of the g-code where the brim is printed. Instead of exporting values for E, only zeroes are exported. But E is omitted. This causes export of non valid g-code. Not only Klipper will throw an error with such g-code i assume, i am certain any other firmware will do too.

I also observed that in combination with spiral vase, the is NO MOVEMENT generated at all.
After the falsy brim g-code i see something like this:
```g-code
;LAYER_CHANGE
;Z:12.8
;HEIGHT:0.2
G1 Z12.600 F11100.000
;### AFTER_LAYER_CHANGE
M117 Layer 63 (Z=12.8)
G1 F1584.000
;LAYER_CHANGE
;Z:13
;HEIGHT:0.2
G1 Z12.800 F11100.000
;### AFTER_LAYER_CHANGE
M117 Layer 64 (Z=13)
G1 F1584.000
```
Only layerchanges will be exported, without any movement in between.
Also the estimated printing time for that particular file with spiral vase and no extrusion combined is estimated at 37 seconds.
The same object sliced with no extrusion but marlin flavor is estimated at 15 minutes.

Here are 3 different g-code files: [g-code.zip](https://github.com/prusa3d/PrusaSlicer/files/5967278/g-code.zip)
- ***Shape-Cylinder_NoExtrusion_Vasemode***
This file will show you the faulty g-code export i am referring to in this issue right now. It was sliced with the no extrusion g-code flavor and bein in vasemode aka spiral vase.

- ***Shape-Cylinder_NoExtrusion_NoVasemode***
This file is for comparison, showing that slicing the same object without vasemode aka spiral vase **but** using no extrusion will export the movements, but also exports faulty g-code.

- ***Shape-Cylinder_Marlin_Vasemode***
This file is for comparison, showing that slicing the same object using marlin as g-code flavor and vasemode aka spiral vase exports working g-code.

The object which is sliced in this case is a simple shape cylinder, generated within PrusaSlicer itself:
[image]

rtyr, 2021-02-12:
I was able to reproduce it.

th33xitus, 2021-08-07:
This issue is still present in PrusaSlicer 2.3.3

bubnikv, 2021-08-09:
The "No extrusion" firmware flavor was never complete.
I have tried to fix it with 9f9cbb46f9415b052db9de0f22ee08ba475c418b
I hope the E axis is no more emitted, however the G-code viewer works no more as there are no extrusion moves produced.

IMHO the "No extrusion" firmware flavor was not a good idea to implement in the first place. What is your use case for it?

th33xitus, 2021-08-09:
> I have tried to fix it with 9f9cbb4

Thank you. If i find the time, i will compile the master branch myself and can give feedback here.

> however the G-code viewer works no more as there are no extrusion moves produced.

That issue would have been a future feature request of me. Without closer looking at how the code really works, i assume you relay soley on the extrusion moves instead of G1 moves. Which is a pitty in that case. Maybe there could be an exception be made in case there are no extrusion moves found in the gcode. So that there are some sane default values taken into consideration for "virtual" extrusion moves just for the sake of displaying some lines for the movements.

> What is your use case for it?

The initial usecase which led to this issue post was, that i was stress testing my klipper firmware and its host-system to find out at which speed the firmware shuts down, creates stutters or generally behaves strange. And i wanted to make those tests as a dry-run. So i had to remove all extrusion moves and knew that PS has this "No extrusion" flavor. Of course that would be possible with a post-processing script too and some regex magic, but yeah. Why not using a function which is inbuilt.

In the meantime though, i saw a support question on the klipper discord that someone was asking what's the best way to slice gcode for a plotter. he converted his 3D-printer to a plotting machine basically and only needs x, y and limited z moves. this question dragged me back to this issue, wondering if it got fixed in the meantime. I assume that plotter like machines probably don't want to be targeted by PS but maybe there are some more usecases for a working no extrusion flavor which i overlooked now.

lukasmatena, 2021-11-08:
The original issue should be solved? Can we close this?

th33xitus, 2021-11-08:
> The original issue should be solved? Can we close this?

Yes, sorry i totally forgot to report back.
The g-code is now valid. The g-code viewer is not working in that case though since its tied to extrusion moves.

lukasmatena, 2021-11-09:
@th33xitus Thanks. I'll close the issue then. I'm not sure how to deal with the "G-Code Viewer shows no extrusions when there are no extrusions" problem (although I understand there may be valid use cases). If you think it is worth considering, please, open a new issue.

closed 2021-11-09 by lukasmatena