prusa3d/PrusaSlicer #6024: Mac OS build issues

bakaufman, 2021-02-12:
### Version
Master

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
Build
### Operating system type + version
Mac OS X 10.15.7

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
Irrelevant 

### Behavior
Following the description on: https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md

No problem on deps.
Make -jN fails. 

If I cannot build the master then is there a version of 2.3.0 or nightlies that I can build? Seems like just an oversight specific to Mac build deps possibly. Happy to execute specifics to provide troubleshooting.
[20210211 PS bkauf@BrettsHomeiMac build % make -j2.zip](https://github.com/prusa3d/PrusaSlicer/files/5969518/20210211.PS.bkauf%40BrettsHomeiMac.build.make.-j2.zip)

bakaufman, 2021-02-12:
I just tried the brew based install instructions found here: https://github.com/prusa3d/PrusaSlicer/commit/f2f5632c3ebc3ac2bbe47af60da60d97dc090202#diff-511cc22c45b08ce7ff4c12fdb42d87e164864663150a75f36f5f5966602364e2R82

Errors with the same issue (see attached)
[20210211 brew make bkauf@BrettsHomeiMac build % make.zip](https://github.com/prusa3d/PrusaSlicer/files/5969640/20210211.brew.make.bkauf%40BrettsHomeiMac.build.make.zip)

bakaufman, 2021-02-12:
[CMakeOutput.log](https://github.com/prusa3d/PrusaSlicer/files/5974662/CMakeOutput.log)
[CMakeCache.txt](https://github.com/prusa3d/PrusaSlicer/files/5974663/CMakeCache.txt)

xarbit, 2021-02-20:
I cannot build on macOS BigSur (11.2.1 20D74) either. Seems something funky with cmake..

- cmake version 3.19.5

fails @ 
[ 74%] Completed 'dep_wxWidgets'
[ 74%] Built target dep_wxWidgets

```
-- Installing: /Volumes/Extreme SSD/source/PrusaSlicer/deps/build/destdir/usr/local/lib/wx/config/osx_cocoa-unicode-static-3.1
-- Up-to-date: /Volumes/Extreme SSD/source/PrusaSlicer/deps/build/destdir/usr/local/bin
CMake Error: cmake version 3.19.5
Usage: /opt/homebrew/Cellar/cmake/3.19.5/bin/cmake -E <command> [arguments...]
Available commands:
....
....
```

```
-- Installing: /Volumes/Extreme SSD/source/PrusaSlicer/deps/build/destdir/usr/local/bin/wxrc
CMake Error: cmake version 3.19.5
Usage: /opt/homebrew/Cellar/cmake/3.19.5/bin/cmake -E <command> [arguments...]
Available commands:
....
....
```

dlueder, 2021-04-07:
So I just compiled everything without any issues on an M1 Mac. 

- cmake 3.20.0 (arm64) installed via brew 
- Xcode 12.5 Beta 3 (12E5244e)

Compiled without any issues and runs great under Big Sur 11.2.3.

bakaufman, 2021-04-07:
Maybe clean OS installs will help others then…

bubnikv, 2021-04-14:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/5754

closed 2026-09-29 by prusaslicer-bot[bot]