prusa3d/PrusaSlicer #6025: Why not a first layer (only) extrusion multiplier setting?

rw3iss, 2021-02-12:
I know I'm going to hear some pushback from this, but I'll argue it.

I know the idea is that we are supposed to set the (overall) extrusion multiplier, and then just set the extrusion width for the first layer, and adjust the first-layer/z-index height from the machine to get a better "push"+adhesion for the first layer...

This is alright, but most of the time I just want to extrude a lot more on the first layer only, to both guarantee bed adhesion, and also get a better bottom surface finish on a textured bed. Cura has this feature, and the prints stick and come out a lot better! Besides that, we shouldn't have to adjust three settings (extrusion multiplier, first layer width, and the z-height on the machine), for a simple setting to just want to get more plastic out of the first layer only. It's also just a nice feature to have in general, I think, and I see other people asking for it.

Would it be possible to add such a setting? Or otherwise maybe there is some way to manually add the G-CODE, I'm sure... and then change it again after the first layer. Is there a specific G-Code command that I can search for to check where the z-height changes after the first layer, and then insert the code to change the multiplier there?

th33xitus, 2021-02-13:
> Or otherwise maybe there is some way to manually add the G-CODE, I'm sure... and then change it again after the first layer.

Depending on you firmware, M221 lets you set the flow percentage.
With a simple condition you can check for the first layer. An example for changing the flow to 110% only for the first layer and reset it back to 100% as soon as the second layer starts can look like this:
```g-code
;Changes the flow to 110% only for the first layer
;and reverts back to 100% on the second layer
{if layer_num == 0}M221 S110
{elsif layer_num == 1}M221 S100
{endif}
```

> Also, it's not clear how the extrustion width is actually derived from the nozzle diameter.

Given a nozzle diameter of 0.4mm
extrusion width = 1.125 x nozzle diameter = 1.125 x 0.4mm = 0.45mm

Personally, what i find more confusing and not clear, is the computation over layer height if expressed as percentage.

rw3iss, 2021-02-13:
Righteous! Thank you for the G-Code logic, @th33xitus! 
Your logic regarding the width makes sense. I think I was confused because I thought the sentence I highlighted was referring to the "if not 0" case... where it would use the set value in that field, but re-reading it it's actually saying "if default isn't set AND this value isn't set... then it uses nozzle diameter x1.125". Anyway, thanks.

fe60, 2021-02-23:
One of the options requested in #3635 a year ago.

neophyl, 2022-12-07:
You are using percentages.  When expressed as a percentage then the line width is NOT based on the nozzle size but on the configured LAYER HEIGHT.  As alluded to by th33xitus.

Its one of the most counter intuitive settings in there but its one of those historical ones that if changed can potentially break existing profiles so no one wants to touch it.  Which is why I always set specific width values and do not use percentages in PS.

11iq, 2023-07-31:
I just came across this thread. This can be achieved directly in Prusaslicer: ```Printer Settings -> Custom G-code -> Before layer change G-code```
Add this line: 
```M221 S{if layer_num==0}100{else}95{endif}```
First layer will be printed with multiplier 1, all next layers with 0.95. Set according to your values.

closed 2026-09-29 by prusaslicer-bot[bot]