prusa3d/PrusaSlicer #6027: Individual Objects / Instancing (one at a time) printing doesn't work

rw3iss, 2021-02-12:
### Version
PrusaSlicer 2.30 Ubuntu 20.10

Hi, it seems that if I try to do instance printing (complete individual objects before moving to the next), the printer doesn't seem to care. It still prints them each at the same time, per layer.

Here are my settings:
[image]

Here are the instances I have laid out:
[image]

Do you see anything wrong with this?

rtyr, 2021-02-12:
Please provide your 3mf project file.

rw3iss, 2021-02-12:
Here you are:
https://drive.google.com/file/d/124sYdVQHymzSbCxkmaeLyRjjAasIVzN8/view?usp=sharing

Thanks.

combolek, 2021-02-13:
FWIW, I cannot reproduce it with the provided project file. Seems to work fine for me.
But even in your screenshot if you look at the Z-height numbers on the vertical scale to the right, the numbers go up to 2.70 and then start again from zero four times, as expected. Are  you sure it is not working? What do you see when you move the vertical slider? Maybe the problem is not in slicing?

rtyr, 2021-02-13:
I don't see any problem, it works as expected with your project file.

rw3iss, 2021-02-13:
Thanks... well I take it back then, it does print them individually, but only AFTER the brim. Sorry, I didn't actually let it get to printing - I thought it was doing them sequentially because of this. It prints each brim sequentially, then proceeds with the individual prints. 
Is there a good argument for doing that? I would think it should print the brim at the same time as printing the individual instances... for better adhesion and whatnot, and also so the user can appropriately stick glue down or anything underneath the surface, likewise for better adhesion.
Anyway, feel free to close this issue, if that's by design...

rtyr, 2021-02-13:
You are right about the brim/skirt not being printed sequentially. Not ideal, but it is known issue https://github.com/prusa3d/PrusaSlicer/issues/292. Closing.

closed 2021-02-13 by rtyr

rw3iss, 2021-02-13:
K, thanks for clarifying and looking into it. Have a nice day :)