prusa3d/PrusaSlicer #6029: 3DBed Regression

pmjdebruijn, 2021-02-12:
### Version
`2.3.0~248+g9c568a543`

### Operating system type + version
Xubuntu 20.04

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 (BLTouch)

### Behavior
Just add the printer.
Before 2e55898d78ddca4753adb60c9222c52bc9ee8ba8:
[image]
After 2e55898d78ddca4753adb60c9222c52bc9ee8ba8:
[image]

pmjdebruijn, 2021-02-12:
This seems to affect most Creality printers. The relevant bed texture render fine in any SVG viewer.

Presumably there some rounding or off-by-one error going on?

bubnikv, 2021-02-15:
> renders fine in any SVG viewer.

The print bed triangulation algorithm was changed
in 2e55898d78ddca4753adb60c9222c52bc9ee8ba8 thus the last commit that used
previous algorithm was 820c18923b4543cd8f878ba4f0071c1f73708347. I will
test.

bubnikv, 2021-02-15:
Fixed with a6bd2d76648fef7d2904095b2ebd479f0205c33a
Thanks for heads up.

closed 2021-02-15 by bubnikv