prusa3d/PrusaSlicer #6035: Outer wall (perimiter) is loose when printing an enclosure with walls of 2mm thick on a Ender 3 printer

dirkleeu, 2021-02-13:
### Version
Version 2.2.0 linux-x64
### 3D printer brand / version + firmware version (if known)
Ender 3 
Marlin 1.1.6 with BL Touch V3.1

### Behavior
The outer wall (perimeter) is loose when printing with PLA.  I did the following to confirm if it is the printer:

1) replaced the hot end
2) slowed the speed on the outer perimeter to 15mm/s
3) printed at 200 degrees PLA
4) Thin wall compensation on and off

I have put the same model in Cura 4.8.  Printed perfectly.  I have attached all the files that I think is relevant.

[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/5975128/PrusaSlicer.zip)

[cardholder_V07.00.zip](https://github.com/prusa3d/PrusaSlicer/files/5975130/cardholder_V07.00.zip)

* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

pmjdebruijn, 2021-02-13:
Is this reproducable with PrusaSlicer 2.3? The creality profiles have seen some changes since then... (Generic PLA has been raised to 205C by default for example).

Please do any further testing/feedback with PrusaSlicer 2.3

I also noticed you're printing at a layerheight of 0.24, do you also have this issue when printing at for example at 0.16?

Do you still have the stock extruder on the Ender 3? As it's extremely prone to underextrusion due to it's weak extruder design... And the underextrusion tends to be variable depending on the filament and print temperature.

Have you printed a temp tower for your particular filament: https://www.thingiverse.com/thing:2615842 ?
While 200/205C seems most common for most PLAs, some PLAs really need 210/215 to flow properly.

Once you have the proper temperature dailed in, you should do a flow rate calibration (to compensate for underextrusion), and it's not unlikely you'd end up at something like 1.03 (or 95/96 E-steps).

dirkleeu, 2021-02-14:
Hi Pascal,

Thanks for your response.  I have verified all of the above.  I was
actually printing during tests at 210C also.  I have been printing with the
original nozzle, but replaced the entire hot end and nozzle with the same
results.  I also did a flow rate calibration.  I even went so far to
increase the flow rate to 1.05.  The reason why I uploaded the report is,
when I did the same with Cura 4.8, I had no problem.  That is why I came to
the conclusion that there must be something different in the way that gcode
is generated.

I am very happy with the PrusaSlicer and prefer it, because it is much more
intuitive, but for the moment I will have to use Cura until I can afford a
Prusa printer.

Yours sincerely
Dirk

pmjdebruijn, 2021-02-15:
Can you post a photo illustrating the issue you're seeing...

dirkleeu, 2021-02-15:
Dear Pascal,

See the attached pdf.  I hope this helps.

Yours sincerely
Dirk

pmjdebruijn, 2021-02-15:
I don't think GitHub transparently handles email attachments in email replies.

dirkleeu, 2021-02-15:
Didn't know, Ok, I will cut-and-paste the info into the mail:

The screwdriver shows how the outer shell did not attach to the rest of the
enclosure

[image: 3d print box incorrect.jpg]

In the below drawing, the area in green never gives a problem, but the area
where the arrow is pointing also gives a problem. A “roof” for the
enclosure partially shown on the left also does not give a problem.
[image: 3print problem drawing.png]Thanks for all your help

Yours sincerely
Dirk

dirkleeu, 2021-02-15:
Hi Pascal,

I just saw that my photo's are not showing.  I have attached them to this writing.  Hopefully it will allow you to open it.
[image]
[image]

pmjdebruijn, 2021-02-15:
With the black filament it's really hard to tell what's going on... But if I understand correctly the outer perimeters on the inside of the box of poorly formed? Presumably he outside of the box feels smooth(ish)? But do the inside walls feel significantly more rough?

And you could reproduce this using PrusaSlicer 2.3 using the new 0.16mm profile, right? As 0.24mm might be more sensitive to poor layer adhesion... Which may be a contributing factor...

For completeness sake, what brand/type of filament are you using?

dirkleeu, 2021-02-16:
Dear Pascal,

Both sides suffer from the problem, it is on the outside and the inside.  I
have not tested this on the 0.16mm profile.  When the vertical area is a
large surface the problem occurs.  Both sides are very smooth as expected
and on both the outside and the inside in the vertical direction the layers
binded surprisingly strongly given that it is only one layer thick.  When I
remove the outer layer the second layer inside is firm and feels
significantly more rough.

The reason why I reported the issue is I realised when I use Cura 4.8, I do
not have the problem.  It means that either Prusa is slicing differently or
I made some mistakes on the Prusa settings, because both print with a 0.24
mm setting, Although you might be onto something because Cura was actually
0.2 mm!

I will verify tonight.

Yours sincerely
Dirk

dirkleeu, 2021-02-16:
Dear Pascal,

I forgot to mention that  the filament brand is CCTree.  At the moment
there is a huge shortage in PLA in South Africa, so you grab what  you can
get.

Yours sincerely
Dirk

pmjdebruijn, 2021-02-16:
With regard to the PLA, good, I just wanted to verify it's nothing exotic.

Another thing I was wondering about, is this the only model you've experience this issue with?

As a point of reference, I was able to print the attached simple model just fine at 0.24mm (using PrusaSlicer 2.3):
[semiopencube.zip](https://github.com/prusa3d/PrusaSlicer/files/5989949/semiopencube.zip)

dirkleeu, 2021-02-16:
Dear Pascal,

I can confirm that all my enclosures suffer from the same problem.  By the
way, the screwdriver shown in my previous photo was actually inserted
between the outer layer and the enclosure wall to show that the outer wall
is loose.  It is not loose at the corners.  I have looked at the attachment
and it is exactly the kind of print that caused trouble on my machine.  I
am printing quite a large finger guard for a fan at the moment so I will
not be able to verify the change tonight.
I will also check the upgrade as I am currently running on Prusa 2.2.

Thanks for all your advice!

Yours sincerely
Dirk van der Merwe

jeloneal, 2025-01-24:
Sorry to pick up that old issue but my prints suffer from the same behaviour on some prints as well. PETG and PLA affected, brand does not matter (Prusament or Creality etc.). My printer is a Prusa Mini+ unmodded and using the newest 2.9 version of the slicer.

It affects not only straight walls but also rounded ones. My oberservations are that if you stick with non arachne algorithm it seesm to work better, but you get the according downsides as well then.

Interestingly, if it happens, it affects the whole wall in that area except the very first layers. So you can almost peel of the wall effortlessly.

Already tried different extrusion factors etc. but it does not really change without negatively affecting other parts of the print. I tend to blame this on an flaw with Arachne algorithm and it's way of filling gaps.

closed 2026-09-29 by prusaslicer-bot[bot]