prusa3d/PrusaSlicer #6036: Another Linux wxwidgets 3.0 issue

clytle374, 2021-02-13:
Gentoo Linux, having issue building with. Hopefully 3.2 will become stable soon. 
```
git describe --tag
version_2.3.0-248-g9c568a543
```
Failure is as follows```

[ 95%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/ExtraRenderers.cpp.o
/home/cory/sources/prusa/src/slic3r/GUI/UnsavedChangesDialog.cpp: In member function ‘void Slic3r::GUI::ModelNode::UpdateIcons()’:
/home/cory/sources/prusa/src/slic3r/GUI/UnsavedChangesDialog.cpp:204:47: error: operands to ?: have different types ‘wxString’ and ‘const char*’
  204 |         m_old_color_bmp = get_bitmap(m_toggle ? m_old_color : grey.c_str());
      |                                      ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cory/sources/prusa/src/slic3r/GUI/UnsavedChangesDialog.cpp:204:47: note:   and each type can be converted to the other
/home/cory/sources/prusa/src/slic3r/GUI/UnsavedChangesDialog.cpp:206:47: error: operands to ?: have different types ‘wxString’ and ‘const char*’
  206 |         m_new_color_bmp = get_bitmap(m_toggle ? m_new_color : grey.c_str());
      |                                      ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/cory/sources/prusa/src/slic3r/GUI/UnsavedChangesDialog.cpp:206:47: note:   and each type can be converted to the other
```

Thanks, Cory

pmjdebruijn, 2021-02-13:
Fixed in #6028

lukasmatena, 2021-02-15:
Fixed by merging #6028, thanks @pmjdebruijn.
Closing.

closed 2021-02-15 by lukasmatena