prusa3d/PrusaSlicer #6040: 3mf import error

devrosx, 2021-02-13:
### Version
_Version of PrusaSlicer used goes here_
2.3.0+

### Operating system type + version
OSX 10.14.6

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S with MMU2S

### Behavior
cant import this 3MF file (exported from Blender)
PrusaSlicer warns about model size and then imports empty model...
Model is not visible with and without 

_Is this a new feature request?_
no

[napis2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5976228/napis2.3mf.zip)

combolek, 2021-02-13:
This seems to be fixed or at least improved after the 2.3.0 release. As a workaround, can you try this?
1. Answer "Yes" when it asks about conversion from inches
2. Click "OK" to the warning about size
3. Click the "Arrange" button in the toolbar (not the "A" key)
4. Scale to the size you need
[image]

devrosx, 2021-02-13:
Thanks workaround works... cant test current version

thanks

closed 2021-02-13 by devrosx