prusa3d/PrusaSlicer #6048: How to make support generation smarter

batata004, 2021-02-14:
By far PrusaSlicer is the best slicer I used in years. By a long shot. But there is one thing that still holds me down to other slicers and I am sure many other people have the same feeling judging by the amount of people complaining about support generation of PS. I am not complaining about adding features to support generation, but its generation itself is really bad. 

Take a look at this image below:

[image]

First is the zigzag at the 1st layer. Completely unnecessary. I use lots of other slicers and I have absolute zero problems with them regarding support, and they never use this zigzag. I know there is already a feature request 1 year old to allow users to disable this zigzag, but it's still not implemented (I am not complaining that it was not implemented, this is open software, and I should be able to do that myself. But this intro just proves my point that PS is not focused on providing good support generation, unfortunately).

The other problem that I have is the red arrow below:

[image]

Makes completely no sense for PS to generate that green wall (right side of the arrow) - it does not support anything, it does not provide stable structure to the supports itself, it does not support anything of the part being 3d printed... absolutely nothing. And it wastes material for a long Z height, I have no idea why it goes that tall. Same happens in the image bellow:

[image]

These unnecessary supports (that dont actually support anything) could be used to support many overhangs that are not well supported by support generation. See image below from before and after the layer that is overhanged:

[image]

[image]

You can see that most of the last layer printed above is not well supported by supports. **I know I can increase support resolution in PS in order to make it to create even a completely solid support layer but it's just wasted material**. If the support generation itself could put the support pillars/lines on better places, avoiding creating tall pillars that is supporting "nothing"... the support generation would be much more efficient. Check the image below, all the parts in red (that I painted with arrows and hand drawing) are wasted material, wasted energy, wasted time and all the red arrows below have support structures that are very tall (not only single lines on the base) and are supporting nothing.

[image]

You could tell me: "but those unnecessary support lines are not that unnecessary, they are needed to better support the support itself". I would say "No". Those lines are completely unnecessary as S3D proves to you when generating support. I can assure you if you remove all those external support lines and put right under the part itself, the support structure will not fail, not bend, not fall... It will work. 

So far, the best support I can find is on S3D. It creates supports only below the part, avoiding printing support out of the part itself and it uses a "zig zag" pattern from the bottom untill the top so that many  "small bridges" are created under the overhangs and only under the overhangs.

I know this is open source software and I have no right to complain about anything. I am not complaining. I am giving my point of view of the only thing that really makes me not like this software 100%. I already like it a lot, but support generationg is really bad, sadly. If you dont wanna fix this, no problem! I will keep using S3D to print 3d parts that need support and it will not make a huge impact on my life. But if you are looking to create a better software, this post helps you with a little bit.

Co0olboi, 2021-02-14:
[]()

bttguy, 2021-02-14:
Please consider this FR, +1

scotfor, 2021-02-15:
It is astounding to me that support parameters and behaviour are so bad in PS.
Please fix it.

bubnikv, 2021-02-16:
@batata004 I am challenging you to provide couple of side by side examples of supports: PrusaSlicer 3MF vs. S3D project.

There are some differences between PrusaSlicer and S3D. Namely PrusaSlicer will not support bridges if "don't support bridges" is enabled and S3D will generate insufficient supports with automatic support generator. It may seem S3D is smart by providing just the right amount of supports, but it is really just blind: It samples the mesh at a regular grid and if your overhang does not fall into the center of a grid cell, than you will not get any support generated. PrusaSlicer on the other side is on the heavy side, putting rather more supports than necessary to not miss small islands. I am not claiming that PrusaSlicer is super smart, I am just claiming that PrusaSlicer is not worse than S3D. It is just different.

I have marked your images.

Image 1, 2, 3: The side of the object is being supported by a zig/zag line. I agree that the "zag" line is not needed. However if S3D placed the support there, it would make exactly the same structure. It just does not support the sides, so the islands are often not well supported.

Image 4: This seems to be a bug. What PrusaSlicer are you running? I think we had a bug in PrusaSlicer 2.2, that did not support bridging perimeters, but this is weird. I would like to examine this issue. Please provide a 3MF.

scotfor, 2021-02-16:
@bubnikv Here a couple of screen snips. One from a cura based slicer, one from PS

Cura
[image]

PS
[image]
 No brim set for the PS.

Generally the PS is a lot messier.

batata004, 2021-02-17:
@scotfor thanks for providing your image. Your image proves exactly what I am telling here: use zig zag pillars/lines on support and it will be great. Also, the support lines on your part are very good, comparable to S3D.

@bubnikv I really value your comment but considering that PS is so terrible working with supports, I am pretty sure I dont even need to provide you any printacreen of supports on S3D so you can compare with PS.

Just look the prints I gave you on #1 - you can see on my images the amount of red arrows indicating obvious problems. **There are so many unnecessary support lines supporting absolutely nothing that I think it's completely unnecessary of me providing printscreens comparing S3D vs PS.** You can already see literaly dozens of bad generated supports on those prints.

Dont take me wrong: PS is way better than S3D in 99% of aspects. It really does. And if the community wont fix the buggy support generated by PS - I am ok with that! I can keep using S3D when my parts need support since I already paid a license for that.

And after many years taking 3D printing serious I can assure you one thing: when dealing with supports, make them zig-zag. **NEVER EVER align a support line with a printing line on the part - it will most probably fuse together even if I use a big separation value** - and if a bigger value is used, the support will do absolutely nothing. Try to always make  support lines perpendicular to the lines being print on the part (as much as possible) and it will be much much much easier to remove. Those zig zag lines act as small bridges all over the support - it's very effective done this way.

jon-le0234, 2021-03-17:
As a start, some of the default values for the support material could be changed to:
- 0.25 mm z distance
- 75% xy distance
- 3 mm support spacing
- 4 interface layers

Values taken from this support forum post: https://forum.prusaprinters.org/forum/prusaslicer/prusaslicer-hard-to-remove-supports/#post-188350

Sineos, 2021-03-18:
> And after many years taking 3D printing serious I can assure you one thing: when dealing with supports, make them zig-zag. **NEVER EVER align a support line with a printing line on the part - it [...]

Fully agree. When using supports in PS I usually turn them in a 45° angle with a pattern spacing of 1.7 mm and omit all the interface layers. For me way better results than the default setting or the forum settings quoted above.
And as I own a S3D license as well, I can confirm that it really excels in supports (and bit less visible seam). Still PS is **so much** better in all other aspects.

batata004, 2021-03-19:
@Sineos that's a great idea turn the support in 45 angle! Thanks a lot and I totally agree with you: PS is so much better than S3D, except on support.

itrmc, 2021-04-13:
Reduction or control of the first layer pattern should be parametric. Let's name it "Build plate interface buffer" and allow the user to control its sizing in millimeters, with 0 meaning disabled. The default should be 1mm.

bubnikv, 2021-10-05:
PrusaSlicer 2.4.0-alpha3 implements snug supports, see
https://github.com/prusa3d/PrusaSlicer/issues/5533
https://github.com/prusa3d/PrusaSlicer/issues/4783
https://github.com/prusa3d/PrusaSlicer/issues/4171

closed 2021-10-05 by bubnikv

scotfor, 2021-10-21:
Thanks to all contributors for the work on supports. It is very very good.