prusa3d/PrusaSlicer #6049: Don't return to previous layer's location with embedded M600 (colorprint)

espr14, 2021-02-14:
### Version
2.3.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
all

### Behavior
* I add color change at layer by using implemented function in PS.
* Once the layers of the 1st color are finished, M600 is called.
* I change filament.
* Head returns to the last location (end of previous layer) and leaves a small amount of new color there.
* Without any extrusion it moves to the start of the next layer and continues printing.

_Is this a new feature request?_
YES

After finishing the layer, it should move to the start position of the next layer and then call M600. This won't cause visible color spill.

Blue dot is left in the top right corner due to M600 called too soon.
[image]
[clock-colorprint.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5977975/clock-colorprint.3mf.zip)

rtyr, 2021-02-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2672. Closing.

closed 2021-02-14 by rtyr

bubnikv, 2021-02-15:
@espr14 pane kolego, to byste si měli vošéfovat ve firmvéru.