prusa3d/PrusaSlicer #6050: Hangs when opening on Windows 10 20H2

killerb77, 2021-02-14:
### Version
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 137,379MB
WorkingSet: 131MB
PrivateBytes: 225MB
Pagefile(peak): 225MB(762MB)
RAM size reserved for the Undo / Redo stack: 1,074MB
RAM size occupied by the Undo / Redo stack: 0MB
OpenGL installation
GL version: 4.6.0 NVIDIA 461.40
Vendor: NVIDIA Corporation
Renderer: GeForce RTX 3090/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### Behavior
Just put together a new pc build.  Ryzen 9 5950X with 128gb ram and RTX 3090
When opening prusaslicer it hangs on the splash screeen for a few minutes.  It's stuck on "Preparing Settings Tabs...

[image]
"

Attached is a screen capture

bubnikv, 2021-02-15:
Do you have any add-ons installed? Sonic studio? What is your motherboard
brand? Do you have any tools installed from that motherboard?

BTW I am running a similar system with AMD Ryzen 9 3900X and GeForce RTX
2080 Ti

killerb77, 2021-02-15:
I am not sure what you mean by add-ons.  I just have stock Prusasclier installed from the standalone installer.  I do not have Sonic studio installed.  My motherboard is X570 AORUS ULTRA (rev. 1.1/1.2) with the latest bios. F33a

The only tools from the motherboard I installed is the AMD chipset drivers but I got the latest version from AMD, 2.13.27.501.

Do you have any suggestions for me to test/try? Is there a log file somewhere?

bubnikv, 2021-02-16:
Would you please try the following for us?
https://github.com/prusa3d/PrusaSlicer/issues/5573#issuecomment-771020727
Will you get a dialog box as this one?
https://github.com/prusa3d/PrusaSlicer/issues/5573#issuecomment-771087216

killerb77, 2021-02-16:
When I run that version I do not get any dialog box.  For some strange reason today when I use prusaslicer the issue has disappeared and it's opening just fine now without any pause or delay.  Not sure why it's working ok now as I have changed nothing. It's started working before trying that different version.

killerb77, 2021-02-19:
It's not fixed all on it's own.  It's still opening slow, but not every time.  It seems random when it decides it wants to open slow.  Running the console version from the command prompt reveals the moment it pauses during startup.  It displays LoadLibraryA: hid.dll then pauses for minute before finally opening.
[image]

bubnikv, 2021-02-19:
hid.dll is a Microsoft library for enumerating and working with USB Human
Interface Devices (keyboards, various mice and 3D controllers). We are
using this library to enumerate 3DConnexion mice.
It seems as if some USB HID device causes this hang ups. Maybe it is some
USB wireless dongle that tries to connect to a mouse or a keyboard?

We used to have such a report before, when we ran the enumeration too
often. I think it was some weird wireless keyboard that caused such havoc.

killerb77, 2021-02-19:
I do not have any wireless devices or dongles.  Just a wired keyboard and mouse.  Corsair K70keyboard and Razer Mamba TE mouse.  I unplugged the keyboard and then Prusaslicer opened instantly with no delay.  So it must be that hid.dll enumeration hangup.

I do not use a 3d mouse.  Is there a setting or a way to disable hid.dll from loading when Prusaslicer starts?

bubnikv, 2021-02-22:
There is currently no configuration for disabling the 3DConnexion detection.

Interestingly I am using Corsair K63 keyboard without any issue. However
there are many reports of similar issues connected to the Corsair
keyboards.

Really slow Windows boot up since buying K70 & M65 - The Corsair User Forums
<https://forum.corsair.com/forums/showthread.php?t=133976>

(2) Very slow boot (xhci_hcd related)? : archlinux (reddit.com)
<https://www.reddit.com/r/archlinux/comments/4ninph/very_slow_boot_xhci_hcd_related/d45je6o/?utm_source=reddit&utm_medium=web2x&context=3>
> I have a Corsair K70 RGB and one of the endpoints doesn't respond
properly when it is probed by the kernel and that results in a long boot,
because the kernel has to wait for the probe to timeout to move on.

English Amiga Board - View Single Post - WinUAE slow to start? (abime.net)
<https://eab.abime.net/showpost.php?s=d89dccaa52d2a0c58f7f28529bed86fb&p=1214314&postcount=11>

c++ - DirectInput8 EnumDevices sometimes painfully slow - Stack Overflow
<https://stackoverflow.com/questions/10967795/directinput8-enumdevices-sometimes-painfully-slow>

killerb77, 2021-03-27:
I just wanted to let you know that my issue is now resolved.  It is 100% the Corsair keyboard.  I have replaced it with a Razer keyboard and the issue is completely gone.  Windows actually loads faster now too.  I never realized it hanged on boot waiting for the keyboard to initialize.  The sad thing about Corsair is that they are aware of the issue but won't issue new firmware correcting the problem.  Must be a design flaw on their part.

jventerprises, 2021-06-08:
great, this is not the first time i have had this exact problem with this keyboard.  looks like its time for a new one...

sorren4, 2021-10-03:
> Do you have any add-ons installed? Sonic studio? What is your motherboard brand? Do you have any tools installed from that motherboard? BTW I am running a similar system with AMD Ryzen 9 3900X and [...]

I have recently moved from my Sound Blaster AE-5 Plus to the onboard Sonic Studio based sound card on my Asus ROG X570 mainboard.  I'm not sure what the correlation is here but now PrusaSlicer isn't loading, not even after many minutes of waiting.

bubnikv, 2021-10-04:
@sorren4
> I'm not sure what the correlation is here but now PrusaSlicer isn't loading, not even after many minutes of waiting.

Does PrusaSlicer just crash, or does it close or does it really hang? If it hangs, then you should see it in the project list.

sorren4, 2021-10-04:
> @sorren4
> 
> > I'm not sure what the correlation is here but now PrusaSlicer isn't loading, not even after many minutes of waiting.
> 
> Does PrusaSlicer just crash, or does it close or does it really hang? If it hangs, then you should see it in the project list.

It hangs indefinitely.  Not sure what project list you're referring to.  The outline of the app and some of the window border information are visible but mostly the PrusaSlicer splash screen hangs at the same place as shown at the top of this thread.

bubnikv, 2021-10-05:
Could you please try the alpha3 for us?

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha3

sorren4, 2021-10-05:
> Could you please try the alpha3 for us?
> 
> https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha3

I have the same issue when loading Alpha 3.  However, the popup that comes up before Alpha 3 loads that outlines the potential conflicting software helped clarify what I might try shutting down.  While I shouldn't have to shut those apps down, when I do, even the current GA release loads just fine.  Hopefully these conflicts get resolved before the next GA release.

bubnikv, 2021-12-22:
Issue of @killerb77  was resolved, the keyboard was the issue.
@sorren4 did not respond for couple of months.
@stewwy This seems to be a different issue from the original one. If you still have an issue, please open a new one specific for your issue.
Closing

closed 2021-12-22 by bubnikv

bubnikv, 2022-04-19:
@killerb77 Now this is funny (or maybe not that funny), because my trusty 6 years old Fujitsu laptop started to fail to boot irregularly after some Windows update. I thought the laptop is dying. After two failures to boot Windows decided to do some "repairs" like rolling back some updates. After couple of months I ended up soundless, no way to fix the sound issues. I was ready to thrash the laptop, but first I gave it a backup and fresh reinstall. First it was ok, but after some update the issue was back. At the end of the day I identified it as hanging on the Corsair keyboard :-) connected to an external USB hub. With the keyboard disconnected the computer seem to boot and function with no issue.

jventerprises, 2022-04-19:
Welcome to the keyboard club!