prusa3d/PrusaSlicer #6051: Error in Slicer

billbat1, 2021-02-14:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

billbat1, 2021-02-14:
I get an error with the SL1 adding holes , So if you hollow it first then it works, but if I then add the holes it fails
[HO_LogCabin_Roof_Column_2012-02-14_Rev 4c- SL1 - Copy.txt](https://github.com/prusa3d/PrusaSlicer/files/5978292/HO_LogCabin_Roof_Column_2012-02-14_Rev.4c-.SL1.-.Copy.txt)
Change the files to .3mf

rtyr, 2021-02-14:
Please provide some description.

billbat1, 2021-02-14:
Please review this error

rtyr, 2021-02-14:
> but if I then add the holes it fails

What do you mean by that exactly? Crash?

billbat1, 2021-02-16:
It closed down the prusa slicer, and I had to reopen it

rtyr, 2021-02-16:
OK, can you share some details about your system? I was not able to reproduce it on Windows 10 64bit.

closed 2026-09-29 by prusaslicer-bot[bot]