prusa3d/PrusaSlicer #6053: Bottom perimeters bad with raft

TS232CZ, 2021-02-15:
### Version
Version 2.3.0+win64

### Operating system type + version
Win 10 Pro x64 20H2

### 3D printer brand / version + firmware version (if known)
M3D
Flux Delta +

### Behavior
The problem is long lasting and perhaps inherited from Slic3r long ago.
Since i saw similar or same issue on my two different printers (despite non of them is Prusa printer) i believe it is something in the gcode of the bottom layers despite preview shows it properly. It is related to bottom layers, more specifically bottom layer(s) perimeters. Layers above are not affected.

See attached print (photo from the bottom) and you can easily notice that bottom perimeters are somehow shortcutting instead of following the circular shape. Perhaps it is also related to raft. When no raft is used the bottom perimeters look ok.
I think it is caused by the gap between raft and printed model in order to keep it easily removable from the raft so perhaps some slower print of the perimeter or some similar setting could improve the result.

I think you did a great job beautifying the top layers but i think it is a time to look into the bottom layers too.

I do trust you are the experts and find a reliable solution.
the attached is an example but basically any print with rounded bottom parts is affected.

Thank you, díky

Tomáš

#### Project File (.3MF) where problem occurs

[zaklopka 2.zip](https://github.com/prusa3d/PrusaSlicer/files/5981048/zaklopka.2.zip)

[image]

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/5981248/config.zip)

rtyr, 2021-02-17:
We are reviewing this pull request https://github.com/prusa3d/PrusaSlicer/pull/6003. 

In the meantime, lowering the contact z distance should improve the result.

bubnikv, 2021-02-24:
Fixed with aee136cedd890ecb5072d9c9215a824fa3f365b1

First object layer over raft will no more be printed with bridging flow, thus normal bottom layer pattern will apply.

closed 2021-02-24 by bubnikv