prusa3d/PrusaSlicer #6054: PrusaSlicer Crash/Close on Opening or Soon Thereafter

Hyperson1c, 2021-02-15:
### Version
PrusaSlicer 2.3.0

### Operating system type + version

Windows 10
AMD 6900 XT (May be relevant if this is a GPU driver problem)

### 3D printer brand / version + firmware version (if known)

Prusa Mk3s 3.9.92-3524

### Behavior

When starting up Pursaslicer 2.3.0 the typical loading splash with nefertiti shows up. Then the prusaslicer program shows up. If I click anywhere on the program it closes immediately. If I wait several minutes I can usually click one or two tabs before forced closing. If I ever click on import geometry button it will immediately close. 

#### Project File (.3MF) where problem occurs

Occurs before any .3mf is imported

Hyperson1c, 2021-02-16:
I solved this problem with the following steps:

https://answers.microsoft.com/en-us/windows/forum/all/app-crash-with-ntdlldll/9aa59f80-99a1-4f3c-b2f0-7eb72df05b2a

mr-coolish, 2021-02-21:
I was having this issue when it would crash on start up too. It worked fine initially and then stopped after two updates to windows. 
I had installed windows 10 on a new system and thought everything had been updated. I needed to go through 6 more rounds of updating windows and it was finally fixed... 
My system is fully up to date and everything works fine.

lukasmatena, 2024-02-18:
Closing as outdated (and also probably not PrusaSlicer related at all).

closed 2024-02-18 by lukasmatena